kliment/Printrun #1061: Can't launch pronterface on ubuntu 20.04

kramer04, 2020-05-25:
I follow the guide from here
https://github.com/kliment/Printrun/tree/master#running-from-source
when i launch pronterface with
`python pronterface.py`

have this
`WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (/home/duncan/Printrun/printrun/__init__.py)
WARNING:root:Could not setup DBus for sleep inhibition: No module named 'dbus'
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
    app = PronterApp(False)
  File "/home/duncan/Printrun/printrun/pronterface.py", line 2397, in __init__
    self.mainwindow = PronterWindow(self)
  File "/home/duncan/Printrun/printrun/pronterface.py", line 219, in __init__
    self.reload_ui()
  File "/home/duncan/Printrun/printrun/pronterface.py", line 295, in reload_ui
    self.createGui(self.settings.uimode == _("Compact"),
  File "/home/duncan/Printrun/printrun/gui/__init__.py", line 233, in createGui
    controls_sizer = ControlsSizer(self, controls_panel, mini_mode = mini)
  File "/home/duncan/Printrun/printrun/gui/controls.py", line 390, in __init__
    else: self.make_standard(root, parentpanel, standalone_mode)
  File "/home/duncan/Printrun/printrun/gui/controls.py", line 423, in make_standard
    llts.Add(wx.StaticText(lltspanel, -1, _("XY:")), flag = wx.ALIGN_RIGHT | wx.ALIGN_CENTER_VERTICAL)
wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_RIGHT)" failed at /home/wxpy/wxPython-4.1.0/ext/wxWidgets/src/common/sizer.cpp(2098) in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers
`

volconst, 2020-05-25:
The current source code does not contain wx.ALIGN_RIGHT. Sync your working dir with HEAD

volconst, 2020-05-25:
Duplicates #1051

kramer04, 2020-05-26:
ok will try it

closed 2020-05-26 by kramer04