kliment/Printrun #1065: I Need Assist

LeonaK, 2020-06-12:
I entered a G Code in Pronterface, and it created a runaway scrolling in the readout box.  Turning computer and printer off and on does nothing.  If I click and hold the red X to close the program, it freezes, then starts scrolling again.  It scrolled for 15-20 min before I realized it was in a loop. I tried uninstalling and reinstalling, it starts scrolling immediately.  Can you tell me how to undo this?  klajda@sbcglobal.net.  Thanks!
[image]

volconst, 2020-06-13:
This could be caused, by my recent commits. Are you using the latest from master branch?
Try the state before them https://github.com/kliment/Printrun/tree/1423eb46eb7ee095d1737c5382090b33c61e4b86

rockstorm101, 2020-12-12:
Closing due to lack of feedback. Please feel free to reopen if needed.

closed 2020-12-12 by rockstorm101