kliment/Printrun #1067: 3D clipping wrong on hires display.

ericmarsh, 2020-06-21:
I'm running Kubuntu 20.04 on a Lenovo 720 with 4k display. I've installed the Printrun package on my laptop and have noticed that the grid that represents the 3d printing plate is clipped to about 1/4 of the yellow area intended for that purpose. 

A screenshot can be seen at [https://photos.app.goo.gl/o1NwXVSitiN2AX3k6](url)

rockstorm101, 2020-06-21:
Hi Eric, could you please provide more info like what version of Printrun are you using (I assume is 2.0.0~rc5-1ubuntu2) and whether there is any sort of console output when the issue happens.

Unless I'm missing something, 'pyglet' is set as a dependency for the 'printrun' package. So it should be installed automatically when you installed Printrun in Kubuntu. Could you please elaborate on how exactly 'pyglet' is removed (or not found) and whether that fixes the issue?

ericmarsh, 2020-06-21:
Hi Rock,
Printrun is just 2.0.0 rc5. I had pulled it down using apt but then when I started running into problems I removed it and just did a git clone. 

I spent some time messing with python to get wxPython to get the app to run without throwing exceptions and upgraded at the same time. At this point I've got 3.8.2 running. When I started after sorting that out I received output like this:

**_emarsh@dogbert:~/Applications/Printrun$ python pronterface.py 
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (/home/emarsh/Applications/Printrun/printrun/__init__.py)
WARNING:root:RPC server bound on non-default port 7982
3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/emarsh/Applications/Printrun/printrun/gui/viz.py", line 69, in __init__
    import printrun.gcview
  File "/home/emarsh/Applications/Printrun/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/emarsh/Applications/Printrun/printrun/gl/panel.py", line 25, in <module>
    import pyglet
ModuleNotFoundError: No module named 'pyglet'

3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/emarsh/Applications/Printrun/printrun/gui/viz.py", line 90, in __init__
    import printrun.gcview
  File "/home/emarsh/Applications/Printrun/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/emarsh/Applications/Printrun/printrun/gl/panel.py", line 25, in <module>
    import pyglet
ModuleNotFoundError: No module named 'pyglet'_**

Even though it couldn't find pyglet it does run and the view port for the print job is correctly sized though the control buttons are too big.

No relevant errors in the console though it is sqawking about not being able to find Slic3r.

I did a "pip install pyglet"

Now the grid on the view port is about 25% of the size that it should be and is being clipped when I try to zoom or drag it. The control buttons are still too big. They weren't before so I must have messed something up along the way. Worse case I'll delete the directory and do another clone.

As I mentioned before I don't really know python but I guess that may change. I've got Printrun loaded into Visual Code and have run it under the debugger so I do have a means to try to chase this down.

Thanks

ericmarsh, 2020-06-21:
I see what's happening. The 2d view is working but 3d isn't, which is why removing pyglet made things better.

DivingDuck, 2020-06-21:
You maybe miss: 
python setup.py build_ext --inplace

ericmarsh, 2020-06-21:
I don't recall having done that. I just tried it and it doesn't seem to have made any difference.

volconst, 2020-06-21:
@ericmarsh, pronterface 2.0+ is python3 . You can start it with ./pronterface.py to avoid mistakes.
The clipping problem could be because of display scaling. Launch pronterface and below the log panel, enter "!print(self.GetContentScaleFactor())" and Send. If the output is not 1.0, probably this is the cause. 
In ubuntu the needed package is python3-pyglet. It should be installed with pronterface, but as you run it with python2, it is not found.

rockstorm101, 2021-02-01:
Hi @ericmarsh, I know the issue is quite old. I hope you still use and enjoy Printrun. If so, could could you please confirm whether the latest version works for you?