kliment/Printrun #1068: Printer resets on connect

gjpc, 2020-06-21:
My new Prusa MK3s has introduced me to Pronterface. 

Prusa suggests running all prints from the SD card to get all the features of the firmware.

If I connect Pronterface to my printer while it is printing from the SD card, the machine is reset and print is aborted without a hope to continue. 

Does Pronterface have to reset the printer on connect? After all, there is a reset button right next to the connect button.

kliment, 2020-06-21:
Hey!
There is a setting in Settings -> Options -> Printer Settings called DTR. Disable that and Pronterface will attempt to connect without a reset (if your USB-serial converter driver allows it).

gjpc, 2020-06-21:
Thanks Klment!
An Answer that only produces more questions!

1. Why is Prusa distributing the 2014.03.10 version with their new printers when a 2015 version with the DTR option in Settings->Options->Printer Settings?
2. Why does unchecking Pronterface V 1.6.0's DTR box still reset the printer on connect on Windows 10 with a USB connected Prussa MK3S?
3. Why does unchecking Pronterface V 1.6.0's DTR box reset the printer on connect on macOS 10.14.6 with a USB connected Prussa MK3S?
4. Why is the Application so old when the commits are so new?

kliment, 2020-06-21:
1. I honestly have no idea. Maybe they forgot about it?
2/3. Unsure, it might be a driver issue or a hardware issue. All we can do in software is set the flag on connect and hope the rest of the chain does the right thing. Windows in particular is awful about this and on some devices you get a reset even from plugging in the device without even connecting to it. That said, you should probably see if the latest release fixes it. See next point.
4.  We've moved to github for releases - try this one: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5

gjpc, 2020-06-21:
Hi kliment,
Baffles me too. Looks like the printer bios resets itself whenever a connection is made over serial. 
neophyl on the [Prusa forum](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/why-does-pronterface-cancel-a-print-in-progress-on-connect/#post-224009) gives the following explanation.

> Its part of the firmware update process on the boards.  When a device connects the board resets and checks that port to see if it should go into bootloader mode to update its own firmware. If it detects a programmer on that end it can then update etc.  I'm not even sure its the printer firmware thats doing this,  it may be the bootloader firmware on the board, so a level below the printer itself.
If it didnt do this it would be much more trouble updating firmware.

So I'll be sure that pronterface is attached before doing any printing if I am using it.

closed 2020-08-01 by gjpc