kliment/Printrun #1071: Hello, my printcore did not always be able to connect to serial port!

Heermosi, 2020-07-09:
Given the condition:
0. I'm working on win7, with a x64 cpu (i3)
1. I've bought a new 3d printer
2. the printer VID/PID/REV is 1A86/7523/0263  and I installed driver from wch.cn given the driver version 3.4.2014.8 (newer version cannot work)
3. the printer required to be connected at 250000 bps I set the bps into the connection

What I see:
1. software shipped with the printer was prontface, with 1.6.0 version, not the final version, with python 2.7
2. Prontface cannot alway connect to the printer successfully, some times, when I connect it just not respond anything.
3. If I ran my python with printcore and etc extracted from the prontface, it has a more than 80% chance it cannot re-connect to the serial port after I exited via Ctrl + C. I have written code to do clean up

```
try:
finally:
   printcore.disconnect()
```
4. even if I keep connect to the port, it had some chance for the port to not respond to my instructions, this is also observed on prontface, it might be caused an unresponding write operation. 

`[ERROR] Can't read from printer(disconnected?) SerialException: call to ClearCommError failed`

5. After I disconnected -> plug out -> plug in -> connect, it has some chance to revive.

I'm tired of tracing this problem, can any one tell me how to keep it working?

volconst, 2020-07-09:
Hi,
How did you extract .py file from Printrun-win-18Nov2017.zip ?
Try a newer version, e.g. https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5
or the lastest from the master branch.
Running printcore seems exotic to me. Try pronsole with connect and reset commands.
Remember how fast connect happens when it succeeds, and try pressing reset and connect button several times when connect timeouts.
Your printer control board may have a reset button - press it and click pronterface connect button immediately.
Try a different USB port.
Try a different USB cable.
Try a different OS.
Try a different host software (printrun rivals).
Print using sdcard.

Heermosi, 2020-07-15:
> Hi,
> How did you extract .py file from Printrun-win-18Nov2017.zip ?
> Try a newer version, e.g. https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5
> or the lastest from the master branch. [...]

Thank you very much.
I've identified the problem might be lying in the driver for USB-Serial CH340 chip, it some time failed to communicate with serial port, and cannot revive when re-connected(got stuck at USB device bootup).
After a reset operation(disable COM? then enable COM?), the communication revives some how. If I start printrun before the USB-Serial port is fully loaded, it would also got stuck at the device bootup and cannot respond to any GCode.
It was almost certain that, either the chip CH340 or the driver for it was bugful.

Heermosi, 2020-10-14:
I've identified the problem is caused by a bugful driver provided which CH340 chip which converts serial port to usb.
When some time the serial port is temperorarily cut the driver halted and cannot be reconnected unless you plug off and then plug in again!
Just look out the issue.

closed 2020-10-14 by Heermosi