kliment/Printrun #1072: 2.0.0rc5 - unzip and run but Fatal Error

hyuntoe, 2020-07-10:
after double click the "pronterface-windows-2.0.0rc5.exe"
it shows a dialog box
Fatal Error!
---------------
Failed to execute script pronterface

Clicked "OK" button, the dialog box disappears,
but there are two pronterface process in the Task Manager,

they won't disappear till reboot computer.

I couldn't see anything about 2.0.0rc5...
how can I do?

rockstorm101, 2020-07-10:
Hi @hyuntoe, what Windows version do you run?

@DivingDuck, could it be related to 32 bit vs 64 bit architectures?

DivingDuck, 2020-07-10:
@rockstorm101 it's the original executable from here, right? This won't work with actual Windows versions.
This is related to the official pyinstaller that have a problem with windows. Same issue as the discussion here:https://github.com/kliment/Printrun/issues/1043#issuecomment-616180981. The only way I know to solve this is, that the executable need to be compiled with the 4.0 developer version.

@hyuntoe, are you able to build Printrun from source?

DivingDuck, 2020-07-10:
@hyuntoe, please find[ here](https://github.com/DivingDuck/Printrun) a link to my fork. Here here is a direct link for a actual version of [Pronterface](https://github.com/DivingDuck/Printrun/raw/master/dist/pronterface.exe). Its based on the master branch here w/o additions. Call it a development version. Hope this will work for you. 

You should update this version as soon as the team here will push a new version some when in the near future.

DivingDuck, 2020-12-28:
@rockstorm101, guess this issue can be closed as there was no further answer/question/reaction since five months.

rockstorm101, 2020-12-28:
Agreed. Closing.

closed 2020-12-28 by rockstorm101