kliment/Printrun #1075: Feature request for pause

sledjunk, 2020-07-12:
When we pause a print, can Pronterface send a couple of commands to the printer after pausing.  I would like to see Z raised to max, or at least 50 mm, x to 0 and y to max.  Typiocally, when I pause the print, I have seen something that needs to be corrected, so getting the extruder out of the way and the print accessible would be nice.

Thanks for your efforts.

volconst, 2020-07-13:
Define a macro with the following commands:
pause
!# move z 50
!# G1 x0 y200
!# to move to max_y, max_z use python
!self.p.send_now('G0 X0 Y%d Z%d'%tuple(self.build_dimensions_list[1:3]))

Define a button which invokes a command equal to the name of your macro.
Use the new custom button instead the stock pause button.

sledjunk, 2020-07-13:
Thank you for the information.  I have no experience with using macros but will experiment with this.
Is there a reference documentation for the macro definition?  I am particularly curious about control characters, ie. are the !# characters used to comment out the line for documentation?

volconst, 2020-07-14:
Exclamation mark is prefix for python code and hash is prefix for comment in python code
See https://github.com/kliment/Printrun#using-macros-and-custom-buttons

rockstorm101, 2020-12-12:
Hi @sledjunk did you try @volconst's solution? Does a macro solve your needs? I would consider there's no need for a new feature then and close this issue. Would you agree @VanessaE? Please let us know what you think.

VanessaE, 2020-12-12:
I feel the same as @sledjunk -- a macro that you could make the "Pause" button execute, in the same manner as custom buttons at the bottom of the window, would be a good solution here.