kliment/Printrun #1076: What do Reset, and OFF do?

sledjunk, 2020-07-12:
As asked, what are these buttons supposed to accomplish.  On my printer, there seems to be no response to the reset, other than to stop the print, and the off does nothing noticeable.  This is on a Prusa i3 MkIII, so it is possible this works differently on other printers.
Thanks

volconst, 2020-07-13:
Reset should restart the printer control board as if you unplug and plug the power cord. You may want to reset if the printer seems unresponsive/hung.
'Off' issues gcodes, you should see them in the log and can see their documentation here https://marlinfw.org/meta/gcode/

sledjunk, 2020-07-13:
Thank you for the information.  I can report that on the Prusa i3 MkIII, the reset does not do anything at the printer end.  Whether that is a refusal of the gcode by the prusa firmware or some other issue, I don't know.
Again, thank you for the information as well as the link to data on the gcode format.

volconst, 2020-07-13:
Reset actually toggles the level on the DTR wire of the serial cable. Some boards react to this by restart, some do not probably depending on a solder brigde or jumper.
I am not sure if reset should have visible result, probably it should at the printer display.
Probably reset is to be followed by a connect.
Connect on it's own can toggle DTR depending on a setting. Probably if the setting is Connect to not toggle DTR, then it can be toggled with Reset.
Pronterface is not designed for Prusa printers, there are many firmware configurations, so it is quite normal some buttons to have no effect.

sledjunk, 2020-07-13:
Thanks.  I understand that Pronterface is not designed for Prusa printers, but only mentioned that as information if it was relevant.  If the reset toggles the DTR, then the controller board's reaction to that is the limiting factor.  The Prusa does reset on connect, so a disconnect followed by connect will accomplish the reset.

closed 2020-07-13 by sledjunk