kliment/Printrun #1077: Activity status query - False / True via Command line or Terminal.

Racansky, 2020-07-24:
Activity status query - False / True via Command line or Terminal.

Good day,
I would like to know if there is a function / parameter that could be used to command Pronterface from the Command line (Windops) or Terminal (Linux) or a script to the status of the print activity? All I need is the status true (yes, I print) or false (no, no printing, it is finished).

I would like to integrate this function into my script, which ensures the automatic shutdown of the PC when it is not used on a dedicated graphics card (GPU-load). In order to verify at the same time whether or not Pronterface is running and printing at the moment.

kliment, 2020-07-24:
There is a setting "external commands -> final command" that you can use to make pronterface do something at the end of a print. It can be any command you want. Maybe make it create a file when the print starts and delete it when the print ends, and then check the existence of the file in your script. Would this work for you?

volconst, 2020-07-25:
https://github.com/kliment/Printrun#rpc-server

Racansky, 2020-07-29:
> Existuje nastavení „externí příkazy -> závěrečný příkaz“, které můžete použít k tomu, aby pronterface udělal něco na konci tisku. Může to být jakýkoli příkaz, který chcete. Možná to vytvoří soubor při spuštění tisku a po ukončení tisku jej odstraní a zkontroluje existenci souboru ve skriptu. Fungovalo by to pro vás?

Racansky, 2020-07-29:
> There is a setting "external commands -> final command" that you can use to make pronterface do something at the end of a print. It can be any command you want. Maybe make it create a file when the [...]

Good day,
thank you for answer. I'm afraid what he proposes won't work.
The Prusa i3 MK2 3D printer is connected to the Linux UBUNT 20.04 PC and we will print via your Pronterface.
However, the computer is also used for other functions. I have created an event here in Crontab, a bash script that controls multi-gpu activity (exec nvidia-smi). If the activity is low, the computer shuts down. This is to save energy.
I would need to add a condition to the script asking about the status of the printer.
for example = "If Multi-GPU load is low but Pronterface is in print state = do not shut down PC"
The command at the end of printing would not work, as in another dependency the PC would be switched off = incomplete printing.

closed 2020-07-29 by Racansky

Racansky, 2020-07-29:
> https://github.com/kliment/Printrun#rpc-server

Yes, it looks the right way
I'm sorry, but Python is a BIG unknown to me.
How could this function be put into the condition of a running bash script?
In Pronterface - Option I have RPC Server enabled.

volconst, 2020-07-29:
@Racansky , start/final command should work too
For example:
Start command: touch .printing
Final command: rm .printing
Error command: rm .printing
shutdown.sh:
[ -f .printing ] || is_gpu_used.sh || shutdown

volconst, 2020-07-29:
For RPC way:
create file is-printing.py:
```
#!/usr/bin/python3
import xmlrpc.client
rpc = xmlrpc.client.ServerProxy('http://localhost:7978')
pr = rpc.status()['progress']
is_printing = pr is not None
# shell exit code 0 means no error, and is treated like true by shell ||
shell_semantics = not is_printing
exit(shell_semantics)
```
shutdown.sh:
```
python3 is-printing.py || is_gpu_used.sh || shutdown
```