kliment/Printrun #1079: Display limited to 200x200 grid

VanessaE, 2020-07-25:
Pronterface will only draw a base grid up to 200x200 in the previewer, even if you tell it the printer's limits are much higher.  For example, I'm building a machine with a 320x320 bed, and have created a Pronterface config for it, which has the limits so set.  Yet: 

[image]

These cubes are at the extreme corners of the plate (according to Prusaslicer).

volconst, 2020-07-25:
What is your version? Recent master branch download seems ok, but I did not load a file.
It is possible that your config is broken and default dimensions are used. Did you use Settings->Options menu to set build dimenstions?

DivingDuck, 2020-07-25:
Hi @volconst,
I'm back again and just start check my mails and saw this issue. 
This problem is related to the old version 2.0 RC5. This is fixed with your update #1054, that is also the latest version of the master branch.

volconst, 2020-07-25:
Cool, thank you, @DivingDuck 
Let's see whether @VanessaE confirms fix by upgrade.

VanessaE, 2020-07-25:
That was with commit b94baed63bd189818b7f70af11d53666ca58432f.  And yes, I set the volume dimensions in the Options.

DivingDuck, 2020-07-25:
@volconst ,
Yes indeed, Maybe worth to check  the correct wxPython >=4.1version too
@VanessaE, You need a newer state. It should include this merge [#1063](https://github.com/kliment/Printrun/pull/1063)

VanessaE, 2020-07-25:
Ooops.  Yep, it works fine at HEAD.

closed 2020-07-25 by VanessaE