kliment/Printrun #1083: Resuming after Pause breaks relative E mode

VanessaE, 2020-07-28:
If the printer is in relative E mode as commanded by M83 or G91 in the g-code, and you "Pause" and then immediately "Resume" the print without doing anything else, Pronterface forces the printer back into absolute E mode.

This leads to unpredictable results, but is guaranteed to fail.

As far as I'm aware, there's no way to detect the mode the printer *defaults* to, so I suggest just checking the incoming g-code to see if M83 or G91 are present.  If found, just automatically issue  `M91` followed by `M83` on Resume.  That way Pronterface can do its usual stuff like jogging or homing in absolute mode.

volconst, 2020-07-28:
Some of the specified logic is already implemented. I think only the relative e is not restored on resume. Ideally first could be created gcode test files with the expected result for different combinations.

VanessaE, 2020-07-28:
> Ideally first could be created gcode test files with the expected result for different combinations

In the meantime, here's some g-code as exported from Prusaslicer: [bed spacer-standoff.gcode.zip](https://github.com/kliment/Printrun/files/4991366/bed.spacer-standoff.gcode.zip)

volconst, 2020-07-31:
@VanessaE , you can checkout this branch https://github.com/volconst/Printrun/commits/resume-rel-extr if it fixes the problem
I have created a test gcode ./testfiles/pause-restore-rel-e.gcode . It has to be loaded in Pronterface, connect, print. The gcode pauses by itself and switches verbose logging so you can verify the commands emitted on resume. Test needs the user to press Resume button to continue.
To compare behavior of test before and after fix do:
$ git clone ... or ... git remote && git fetch remote
$ git checkout 5cffab7
$ ./pronterface
$ git checkout resume-rel-extr
$ ./pronterface
After you confirm fix, I will create pull request.

VanessaE, 2020-09-19:
Sorry I took so long to get back to this one.

I tested with an XYZ cube, and it seems to work as it should.  I also used your test gcode file: 

```
Loading file: /home/vanessa/RepRap/Printrun-volconst/testfiles/pause-restore-rel-e.gcode
Loaded /home/vanessa/RepRap/Printrun-volconst/testfiles/pause-restore-rel-e.gcode, 10 lines
0.00mm of filament used in this print
The print goes:
- from 0.00 mm to 0.00 mm in X and is 0.00 mm wide
- from 0.00 mm to 0.00 mm in Y and is 0.00 mm deep
- from 0.00 mm to 0.00 mm in Z and is 0.00 mm high
Estimated duration: 0 layers, 0:00:00
Print started at: 16:42:38
Locking interface.
G-Code calling host command "!print('Please press Resume and check for M 83')"
Please press Resume and check for M 83
G-Code calling host command "!self.p.loud = True"
Print paused at: 16:42:49
SENT: M105
RECV: ok T:239.37 /240.00 B:82.82 /85.00 @:127 B@:127
SENT: M105
RECV: ok T:239.06 /240.00 B:82.60 /85.00 @:0 B@:127
SENT: M105
RECV: ok T:239.06 /240.00 B:82.53 /85.00 @:0 B@:127
SENT: M105
RECV: ok T:238.75 /240.00 B:82.45 /85.00 @:127 B@:127
SENT: M105
RECV: ok T:239.37 /240.00 B:82.41 /85.00 @:0 B@:127
Unlocking interface.
SENT: M105
RECV: ok T:239.37 /240.00 B:82.30 /85.00 @:32 B@:127
Resuming.
SENT: G90
RECV: ok
Print resumed at: 16:43:05
SENT: G1 X0.0 Y0.0 F3000
RECV: ok
SENT: G1 Z0.0 F12000
RECV: ok
SENT: G92 E147.42697999999962
Locking interface.
RECV: X:0.00 Y:0.00 Z:0.00 E:147.43 Count A:36320 B:-4320 Z:8000
RECV: ok
SENT: M83
RECV: ok
SENT: G1 F12000.0
RECV: ok
G-Code calling host command "!self.p.loud = False"
G-Code calling host command "!threading.Timer(2, lambda logbox: print('PASSED. Seen M 83' if 'M' + '83' in logbox.Value else 'FAILED: M 83 not seen'), (self.logbox,)).start()"
Print ended at: 16:43:05 and took 0:00:10
PASSED. Seen M 83
Unlocking interface.
```
I guess that's a success :smiley:

I am not sure what you meant with that last bit about comparing behavior; what I did above suggests no unexpected changes.

volconst, 2020-09-20:
Thank you @VanessaE 
Comparing behavior is about proving that this specific commit fixes the problem.
I see you use [X] Lock checkbox. What's the idea behind it?

VanessaE, 2020-09-20:
Oh you can ignore that, I was testing with the "factory" configuration rather than one of my usual config files and didn't bother to disable that option :smiley:

PR #1114 (2020-09-28): Resume file with relative extrusion
Fixes #1083
closed 2020-09-28 by kliment