kliment/Printrun #1084: Keyboard jogging jumps between widgets

genpfault, 2020-07-30:
This does not work anymore - you can click the circle and get a tiny little D-pad icon, but when you press one of the up or down cursor keys, the input focus jumps from one widget to another (depending on which key you pressed) until it lands on a text field such as the X/Y feed rate or extrude speed, where it sticks (at which point, up/down increments or decrements the value, and left/right move the text cursor in the field).  Pressing right cursor while the focus is on the little D-pad does nothing at first, but holding it down for a second zooms in on the viewport.  Left cursor sometimes moves the X axis, or zooms out, or jumps between widgets, or just does nothing.

_Originally posted by @VanessaE in https://github.com/kliment/Printrun/issues/691#issuecomment-387584895_

genpfault, 2020-07-30:
Moving the jog logic into the `EVT_CHAR_HOOK` handler works for me on Linux as of today's master (82f227d5de):

```
diff --git a/printrun/gui/xybuttons.py b/printrun/gui/xybuttons.py
index 5385171..46a2556 100644
--- a/printrun/gui/xybuttons.py
+++ b/printrun/gui/xybuttons.py
@@ -287,15 +287,13 @@ class XYButtons(BufferedCanvas):
     def OnTopLevelKey(self, evt):
         # Let user press escape on any control, and return focus here
         if evt.GetKeyCode() == wx.WXK_ESCAPE:
+            self.setKeypadIndex(-1)
             self.SetFocus()
-        evt.Skip()
 
-    def OnKey(self, evt):
-        if not self.enabled:
-            return
-        if self.keypad_idx >= 0:
+        if self.enabled and self.keypad_idx >= 0:
             if evt.GetKeyCode() == wx.WXK_TAB:
                 self.setKeypadIndex(self.cycleKeypadIndex())
+                return
             elif evt.GetKeyCode() == wx.WXK_UP:
                 self.quadrant = 1
             elif evt.GetKeyCode() == wx.WXK_DOWN:
@@ -312,14 +310,26 @@ class XYButtons(BufferedCanvas):
                 evt.Skip()
                 return
 
-            self.concentric = self.keypad_idx
+            self.concentric = self.keypad_idx + 1
             x, y, z = self.getMovement()
 
+            if evt.GetModifiers() == wx.MOD_CONTROL:
+                x, y, z = tuple([0.5 * i for i in (x, y, z)])
+            elif evt.GetModifiers() == wx.MOD_SHIFT:
+                x, y, z = tuple([2.0 * i for i in (x, y, z)])
+
             if x != 0 or y != 0 and self.moveCallback:
                 self.moveCallback(x, y)
             if z != 0 and self.zCallback:
                 self.zCallback(z)
-        elif evt.GetKeyCode() == wx.WXK_SPACE:
+            # swallow event
+            return
+        evt.Skip()
+
+    def OnKey(self, evt):
+        if not self.enabled:
+            return
+        if evt.GetKeyCode() == wx.WXK_SPACE:
             self.spacebarCallback()
 
     def OnMotion(self, event):
```

volconst, 2020-09-20:
@genpfault , Thanks for the code, I used some ideas and lines from it for the above commit.
You can check whether it works, it's already in rc7.

genpfault, 2020-10-15:
Looks good, thanks!

closed 2020-10-15 by genpfault