kliment/Printrun #1089: [Ubuntu 20.04] Unable to "Build from Source":

eduncan911, 2020-08-03:
Using `master` branch at the time of writing this issue

```
$ which python
/home/eric/.virtualenvs/printrun/bin/python
$ python -V
Python 3.8.2
$ pip -V
pip 20.2 from /home/eric/.virtualenvs/printrun/lib/python3.8/site-packages/pip (python 3.8)
$ pip install https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/wxPython-4.1.0-cp38-cp38-linux_x86_64.whl
```

Then when trying to install the deps (snipped most of the successful ones):

```
$ pip install -r requirements.txt 
Ignoring pyobjc-framework-Cocoa: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyreadline: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: pyserial>=3.0 in /home/eric/.virtualenvs/printrun/lib/python3.8/site-packages (from -r requirements.txt (line 1)) (3.4)
...
Building wheels for collected packages: dbus-python
  Building wheel for dbus-python (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/eric/.virtualenvs/printrun/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-cfyo98k_/dbus-python/setup.py'"'"'; __file__='"'"'/tmp/pip-install-cfyo98k_/dbus-python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-twnmvyv3
       cwd: /tmp/pip-install-cfyo98k_/dbus-python/
  Complete output (144 lines):
  running bdist_wheel
  running build
  creating /tmp/pip-install-cfyo98k_/dbus-python/build
  creating /tmp/pip-install-cfyo98k_/dbus-python/build/temp.linux-x86_64-3.8
  checking whether make supports nested variables... yes
  checking whether to enable maintainer-specific portions of Makefiles... no
  checking for gcc... gcc
...
  checking for pkg-config... /usr/bin/pkg-config
  checking pkg-config is at least version 0.9.0... yes
  checking for DBUS... no
  configure: error: Package requirements (dbus-1 >= 1.8) were not met:
  
  No package 'dbus-1' found
  
  Consider adjusting the PKG_CONFIG_PATH environment variable if you
  installed software in a non-standard prefix.
  
  Alternatively, you may set the environment variables DBUS_CFLAGS
  and DBUS_LIBS to avoid the need to call pkg-config.
  See the pkg-config man page for more details.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-cfyo98k_/dbus-python/setup.py", line 85, in <module>
      setup(
    File "/home/eric/.virtualenvs/printrun/lib/python3.8/site-packages/setuptools/__init__.py", line 165, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/home/eric/.virtualenvs/printrun/lib/python3.8/site-packages/wheel/bdist_wheel.py", line 223, in run
      self.run_command('build')
    File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-cfyo98k_/dbus-python/setup.py", line 56, in run
      subprocess.check_call([
    File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['/tmp/pip-install-cfyo98k_/dbus-python/configure', '--disable-maintainer-mode', 'PYTHON=/home/eric/.virtualenvs/printrun/bin/python3', '--prefix=/tmp/pip-install-cfyo98k_/dbus-python/build/temp.linux-x86_64-3.8/prefix']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for dbus-python
  Running setup.py clean for dbus-python
Failed to build dbus-python
DEPRECATION: Could not build wheels for dbus-python which do not use PEP 517. pip will fall back to legacy 'setup.py install' for these. pip 21.0 will remove support for this functionality. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
```
[pip-install-error.txt](https://github.com/kliment/Printrun/files/5013331/pip-install-error.txt)

Attached is the entire output.

eduncan911, 2020-08-03:
I should also note that the error and attached error log was from me trying to remove the `requirements.txt` version of `dbus-python3`, to see if it would make a different.

It didn't.  Exact same error, and messages, using version `>= 1.2.0` in the `requirements.txt`.

hroncok, 2020-08-03:
> A possible replacement is to fix the wheel build issue reported above.

Which is:

```
  checking for DBUS... no
  configure: error: Package requirements (dbus-1 >= 1.8) were not met:
  
  No package 'dbus-1' found
```

Try installing (via apt) `libdbus-1-dev`. Does it help?

eduncan911, 2020-08-03:
Installing dev packages on a secure machine is not a solution.

Though yes, I admit that may help.

At this point, I'm building a Docker image to debug this safely.

hroncok, 2020-08-03:
I don't know what a secure machine means, but I'd say installing dev packages maintained by your distro maintainers is a magnitude safer than installing random packages from PyPI.

eduncan911, 2020-08-03:
True.

However, python wheels are isolated to the namespace it runs under - in
this case, a limited user with no sudo rights.

Where as dev packages are available system wide for exploits.

eduncan911, 2020-08-03:
`pydbus` is the preferred lib to use in place of the deprecated `dbus-python`.  

And, it does not require any system-wide dev packages installed.  :)

volconst, 2020-08-03:
@eduncan911 You can try something like this:
```
apt download libdbus-1-dev
dpkg -x libdbus-1*.deb dbus-deb
# show values for DBUS_CFLAGS  DBUS_LIBS
PKG_CONFIG_PATH=dbus-deb/usr/lib/x86_64-linux-gnu/pkgconfig pkg-config  --cflags --libs dbus-1
-I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -ldbus-1
export DBUS_CFLAGS=-I dbus-deb/... -I ...
export DBUS_LIBS=-ldbus-deb/usr/lib/x86_64-linux-gnu/libdbus-1.so
pip install -r requirements.txt
```
Another workaround is to download python3-dbus and set PYTHONPATH
These paths are taken from Ubuntu 19, may be different in your env

Duke-Cardlert, 2020-11-07:
I got similar errors to what was posted above, at the "install -r requirements" step.
Afterward I did apt-get update and upgrade.
I retried the previous step and received the following error message:

"wxPython-4.1.0-cp27-cp27mu-linux_x86_64.whl is not a supported wheel on this platform"

volconst, 2020-11-07:
@Duke-Cardlert , you should use python3:
pip install -r requirements.txt
should be
pip3 install -r requirements.txt

Duke-Cardlert, 2020-11-10:
thank you for the reply. I was able to find success by skipping this
"wxPython wheel" step as shown in the installation guide. If I find that I
need to reinstall, I will do it this way.

all the best!