kliment/Printrun #1102: OS X - cannot connect to serial port

mron, 2020-09-05:
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G103

Printrun 2.0.0rc5

I set up the port and speed
I press connect:

Connecting...
Exception in thread Thread-5:
Traceback (most recent call last):
  File "serial/serialposix.py", line 493, in read
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 260, in _readline
  File "serial/serialposix.py", line 509, in read
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 916, in _bootstrap_inner
  File "threading.py", line 864, in run
  File "printrun/printcore.py", line 345, in _listen
  File "printrun/printcore.py", line 314, in _listen_until_online
  File "printrun/printcore.py", line 281, in _readline
IndexError: tuple index out of range

mron, 2020-09-05:
I tried running from a virtual env. I don't get the errors listed, but Pronterface hangs when I try to connect:

python pronterface.py 
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line'
2020-09-05 13:15:50.217 Python[1091:23885] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to (null)
Connecting...

I can connect to the printer via screen and CNC-GCode-Controller. Slic3r will not connect.

mron, 2020-09-05:
The CNC is running Marlin v2.0

mron, 2020-09-05:
How does pronterface check communication?

volconst, 2020-09-06:
Maybe the serial library is not the correct version.
In the command textbox execute:
!import serial;print(serial.VERSION)
!import serial;print(serial.Serial.port)
!import serial;print(serial.Serial.setPort)
And paste the outputs here.
Execute: python3 -m pip install -r requirements.txt
If you use python for other apps, you may need to setup virtual environment before the above command.
Try getting the today's 2.0.0rc7 and pip install again

mron, 2020-09-06:
Thanks for replying.
I downloaded 2.0.0rc7. I followed directions for running in a virtual env. Here's the output:

>>> !import serial; print(serial.VERSION)
3.4
>>> !import serial;print(serial.Serial.port)
<property object at 0x1183673b8>
>>> !import serial; print(serial.Serial.setPort)
<function SerialBase.setPort at 0x11836dd08>

A couple questions:
1) How does Printrun decide it's connected? I'm running Marlin on a CNC mill. It has no extruders. I added an extruder and used the "998" sensor when I compiled Marlin. With this change Printrun 1.6.0 does connect.

2) Printrun 1.6.0 seems to inject gcode. Is there an option to set the start/stop code?

mron, 2020-09-06:
BTW I tried to change Settings->Options->Extruders Count to 0. Not allowed

volconst, 2020-09-07:
> 1. How does Printrun decide it's connected? I'm running Marlin on a CNC mill. It has no extruders. I added an extruder and used the "998" sensor when I compiled Marlin. With this change Printrun 1.6.0 does connect.
If pronterface reads lines starting with 'start', 'Grbl ' or containing 'ok', or 'T:' it decides it is online.
https://github.com/kliment/Printrun/blob/printrun-1.6.0/printrun/printcore.py#L321
You can enable logging of communications with Settings-> Debug communicatons (if exists in 1.6)
What are the first lines of Marlin2?
Check if the baud rates match after the compilation.
> 2. Printrun 1.6.0 seems to inject gcode. Is there an option to set the start/stop code?
Printrun sends temperature report M105 commands periodically.
In Settings->Options->External commands you can enter OS commands to be executed on print start, stop. You can also define macros with gcodes and create custom buttons with them, but must be clicked to be executed.

Do you still get exception with rc7? Please send the stacktrace for rc7, because I want to look at the line numbers.

mron, 2020-09-07:
How do I set stack trace? I’m happy to help with this.

volconst, 2020-09-07:
The stacktrace (Traceback here) are those 'File...' lines you included in the first post. I just need the ones for rc7 if you still get connect error.

mron, 2020-09-07:
Got it.
rc7 doesn’t dump a stack trace. It just waits for connection. Is the _listen() function used in rc7? If so. I can add a print statement to see if it’s executed.

mron, 2020-09-07:
I’ll poke around with print statements. I’ll start with _listen_until_online(). I’m not familiar with python. I’ll do what I can.

mron, 2020-09-07:
I’m assuming print() will go to the command console.

volconst, 2020-09-07:
Enable logging of communications with Settings-> Debug communicatons and then connect.
Look for RECV lines. If present, the connection is fine, but content is not interpreted correctly.
Paste beginning of log here.

mron, 2020-09-07:
I’m using logging.info()

mron, 2020-09-07:
Here’s the output. No RECV’s. I know it’s using the correct serial port. I get an error if I try to open the same port with screen. There may be an OS permission problem. Since this is an unsigned application, OSX might not allow the use of the port.

I appreciate your help. Am I the only person with this problem?

python pronterface.py
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line'
2020-09-07 08:01:58.551 Python[12400:884775] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to (null)
Connecting...
SENT: M105
SENT: M105
SENT: M105
Not connected to printer.
Disconnecting from printer...
Exiting program. Goodbye!

mron, 2020-09-07:
It has something to do with dtr. I commented out the self.printer.dtr = dtr and the printer connects

                    try:  #this appears not to work on many platforms, so we're going to call it but not care if it fails
                        #self.printer.dtr = dtr
                        self.logError(_("DTR=%s"%  self.printer.dtr) )

I added the self.logError...

volconst, 2020-09-07:
This is great :)
You do not need to change the code, there is a DTR checkbox in Settings->Options->Printer settings.
What is the model of your control board, I think DTR is specific to the model.
Which setting of the DTR works for you - checked or unchecked?

mron, 2020-09-07:
Checked.
Marlin 2.0
Controller SKR1.4

mron, 2020-09-07:
Checkbox? I always do things the hard way.

volconst, 2020-09-07:
> Checkbox? I always do things the hard way.

Me, too. The world ticks, because of us.
I almost believed you, MacUser :)
You also have to reflect on the information that checked DTR is the default state.

mron, 2020-09-07:
In the code yes, not in the UI. I had to check it. Maybe I have an old preference file that I had DTR unchecked sometime ago.
Anyway, it’s working now.
Thanks

volconst, 2020-09-07:
Can you try this:
With DTR unchecked, press Connect and then Reset.
Does it connect?

mron, 2020-09-07:
I unchecked DTR,
closed pronterface,
reopened,
verified that DTR was unchecked,
pressed connect,
no connection,
pressed reset,
no connection,
checked DTR,
pressed connect
connected!!

closed 2020-09-10 by mron