kliment/Printrun #1105: newbie question

sledjunk, 2020-09-06:
This might be a dumb question, but, the last two pre-releases seem to be only source code.  How to I install this on Windows?

Thanks

volconst, 2020-09-07:
https://github.com/kliment/Printrun/pull/1069#issuecomment-687774461

sledjunk, 2020-09-07:
pip is not recognized on my system.  To date, I have only installed printrun from a Windows install.

volconst, 2020-09-08:
You can install python from Microsoft Store https://www.microsoft.com/en-us/p/python-38/9mssztt1n39l?activetab=pivot:overviewtab
or https://www.python.org/downloads/release/python-385/

kliment, 2020-09-10:
I just added a Windows binary to https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7 - please test whether it works for you

closed 2020-09-20 by volconst