kliment/Printrun #1107: Windows 10 virus

Albalate, 2020-09-10:
launches a virus warning and blocks the executable

kliment, 2020-09-10:
What version is this and where did you get the file from?

Albalate, 2020-09-10:
https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc7/Printrun-2.0.0rc7-windows_x64.zip
regards

Albalate, 2020-09-10:
also, Source code

kliment, 2020-09-10:
Okay, I get the same message on W10, and it's the same with any file downloaded from the internet that isn't signed - but you can click "more info" and then "run anyway" and it works correctly.

Albalate, 2020-09-10:
I did that.
but it would be good for people to know in case you doubt
greetings

kliment, 2020-09-10:
Do you know if there is anything I can do about it?

Albalate, 2020-09-10:
sign the executable or place a readme.txt in the zip warning

closed 2020-09-10 by kliment

Albalate, 2020-09-10:
> I don't think signing the executable is going to work (I guess self-signed
> certificates are not allowed and everything else is crazy expensive and/or
> an extreme violation of privacy for the developers) but we will definitely
> add a note on next release. Thanks for your help!

It's Spain we all have our free certificates, I don't know where you are
from and the price

kliment, 2020-09-10:
@Albalate If you know of a source for free codesigning certificates (as opposed to just HTTPS certificates) I'd be very interested - so far the cheapest I found is 69€ starting cost plus 25€/year from a Polish company that wants copies of private documents sent to them which they then sell on to who knows who. Also while this makes the developer name (and only the developer name - not the name of the program) show up in the security dialog, it still doesn't make the scary message go away. Every other certificate issuer is several hundred per year and requires notarized documents sent to them by physical mail. I'm not really comfortable with going down that path, but if you have a better option I would love to know.