kliment/Printrun #1109: Does Printrun filter G Code?

mron, 2020-09-11:
I'm experimenting with a Marlin version that has P000, R000, and S000 as valid gcodes. I'm seeing this response:
*** Unknown syntax: S000
I can't tell if Pronterface or Marlin is complaining. If it's Pronterface, can it allow unknown codes to be passed on?

volconst, 2020-09-11:
Strange, I cant find 'Unknown syntax' neither in printrun's source, nor in Marlin2.x .
Enable Settings->debug communications and if the Unknown syntax is prefixed with RECV, then it's from Marlin.
Paste the whole command that causes the message.
I sent 'M104 S000' and it does not produce error, so it's probably from Marlin.

mron, 2020-09-11:
I could'nt find that message either. The codes I'm testing are literally "P000", "S000", and "R000". No MXXX or GXXX with it.

mron, 2020-09-11:
I'll close this.

mron, 2020-09-12:
I was able to get Pronterface to accept S000,R000 and P000 as valid. The syntax error is coming from Pronterface, but I cannot find the word "syntax" in any error or log message. I think it's coming from python itself.
I don't want to make a fork and pull request for these changes, unless you want them. I would like to know how to make an OS X executable out of my local copy. Running python on a virtual env is a bit of a pain.

volconst, 2020-09-12:
Running from already installed virtual env should be as easy as:
$ . venv/bin/activate
$ ./pronterface.py
Of course it's best to put these in a shell script and just double click it.
You can create executable with pyinstaller, see commands in https://github.com/kliment/Printrun/blob/master/.github/workflows/buildpackage-mac.yml
If you create your own branch and (pull request (to your self)), I think the above commands will be executed and the result could be found somewhere for download.
If you have more questions about the mac build, @kliment could the able to assist and I think you can help us with this problem https://github.com/kliment/Printrun/pull/1069#issuecomment-688854131

volconst, 2020-09-12:
The error message is coming from https://github.com/python/cpython/blob/master/Lib/cmd.py#L236

closed 2020-09-14 by mron