kliment/Printrun #1110: `Shift` + `scroll wheel` does not cycle one layer at a time

VanessaE, 2020-09-15:
Got a minor annoyance here :smile:

Shift+mouse wheel scrolls through layers, but it isn't usually possible to dial right into the layer I want most of the time.

For example, suppose I've just started printing a model, and I want to maximize the model view in the window (bad eyes, so I try to make it easier to see).  Since I have Pronterface set to track the ongoing print, and it'll still be on the first layer in this example, I'll often do `ctrl`+`shift`+`scroll-up` to quickly reveal the whole model, then I'll adjust the view (i.e. pitch/yaw, pan, zoom) to my liking.  Then I'll usually want to wind the display back to layer 1, but I often can't get right on it.

Or to put it more generically, `shift`+`scroll` (either up or down) just usually won't let me dial right in on the layer I want -- it seems to always jump 2 or 3 layers at a time.

`xev` shows the mouse sending exactly one up or down scroll event per "click" of the wheel, so that wouldn't be the problem, and it works fine in other programs.  Even the view zoom in Pronterface seems to move one step at a time in response, just not the scroll-through-layers part.

VanessaE, 2020-09-15:
@volconst a little birdie told me this is your domain :smile:

volconst, 2020-09-18:
Hi @VanessaE , you did not verify #1083

VanessaE, 2020-09-18:
Sorry about that, I haven't had a chance to get to that one.

volconst, 2020-09-19:
@VanessaE , you can download/checkout this branch https://github.com/volconst/Printrun/tree/wheel-duplication and test whether it fixes the problem.

VanessaE, 2020-09-19:
Yep, the mouse wheel seems to work fine in that `wheel-duplication` branch.

closed 2020-09-19 by kliment