kliment/Printrun #1111: Layer detection algorithm

volconst, 2020-09-17:
There is a problem with layer detection https://github.com/kliment/Printrun/pull/1069#issuecomment-692127342 .

> b) Pronterface don't load one of my gcode files correct and tell me that I have 9 layers instead of 265 (pronterfaceRC7GCODE_Error1.PNG). Loading via Plate building tool it seems to be a bit better, now there are 135 layers but still missing 130 layers (pronterfaceRC7GCODE_Error2.PNG). Saving this gcode file ends in an error with an empty gcode file (pronterfaceRC7GCODE_Error2.PNG).
[pronterfacerc7_errors.zip](https://github.com/kliment/Printrun/files/5219265/pronterfacerc7_errors.zip)

The algorithm tries unsuccessfully to extract a single layer height that applies to all layers.
After layer height is determined, non compliant layers are snapped to compliant ones thus reducing the detected layers which is visible when stepping through layers in the 3d viewer.
I attach a small test file based on the original attachment, which also prints the layers z with a host command.

[layer-detect.gcode.txt](https://github.com/kliment/Printrun/files/5238825/layer-detect.gcode.txt)
When printed the attached file outputs there layers [None, 5.0, 0.2, 0.0, 0.503, None]
Proposed changes:
 - Do not create layer if it does not extrude - this will skip layer with z = 5.0
 - Do not detect height and rebase layers to fit the height - this will skip layer @ 0.0,
 - Do not snap to existing layers. Need info (gcode example) why we need to fix almost identical layers. https://github.com/kliment/Printrun/blob/printrun-2.0.0rc7/printrun/gcoder.py#L639
 - Do not create layers with z = None. Maybe except when it is the only one.
To select the correct behavior we need to consider:
 - gcode files with varying layer height. @DivingDuck can you explain why the attached file has varying layer height and how there is layer at Z 0.2 and 0.203 ?
 - where the detected layers are used in pronterface? When stepping though layers with arrows/slider and in layer change callbacks which probably do nothing now, but someone may be using in some way. Are layers used for something else?
@kliment What do you think about the proposed changes?

DivingDuck, 2020-09-17:
I had set up variable layer hight (adaptive slicing) in Slic3r (actual dev branch) for an other part that was more complicated with different layers that had to match in hight. Usually I would not use it for this kind of part. I just recognize I forgot to attach the model too.
@volconst, good catch. You are right, this looks very weird and I haven't a explanation for this. Slic3r add in addition to a brim a skirt.  And this in the wrong order after the brim - what a holly sh**. I sliced this automatically via pronterface and had not recognize this mess. The code is different to what Slic3r showing, there the first layer is printing the skirt (at 0.2) and the second is printing brim and first layer (at 0.203), so there seems to be a weird thing in Slic3r too. 

I start o be frustrated now. First Ultimaker and their update mess with 4.7, Autodesk start yesterday frustrating Fusion360 non commercial user's and now this. It looks like this is not my week
[Z Nut Braket EBFM 8x1,5 R Rotguss AGe neu v3.zip](https://github.com/kliment/Printrun/files/5239349/Z.Nut.Braket.EBFM.8x1.5.R.Rotguss.AGe.neu.v3.zip)

.

kliment, 2020-09-17:
I know some people are using layer change stuff to trigger cameras and such for timelapse photos, but I think breaking that is less problematic than what we have now. The reason we do discrete layers is to be able to scroll through layers in the UI, and the reason we coalesce layers is because otherwise z-lift travel can get counted as a separate layer. I think your proposed changes would make things more robust. There are several other situations where the current layer detection is problematic, not just variable height layers. Spiral vase printing and multi-layer infill generated by some slicers also confuse the layer detection. And of course some advanced toolpath generation will actually follow the contour of the object in Z rather than create flat layers. So I'm on board with relaxing the assumption that layers are the same height. This causes a few problems though:

- How do we deal with it in the UI - users want to incrementally scan through an object, the layer abstraction is an obvious way to do so, but what do we count as a layer?
- Showing travel moves is useful - ideally we'd show travel moves associated with a layer even if they lift in between - this is why merging to closest layer was implemented I think
- If we segment by height, but a toolpath follows the contour of the object, we probably want to group layers together in time rather than in space
- Same thing with variable layer heights

Based on this, I'd propose a slightly different thing - how about we go through the file chronologically and segment it into a new layer when one of a number of conditions is met:

1. E-axis reset (many slicers do this on layer change to prevent E coordinates growing too large)
2. Layer change comment (I think slic3r does this)
3. Step change in Z after XY-only extrusion moves and followed by XY-only extrusion moves at the new Z (this handles the current fixed-height layers)
4. a number of lines (configurable) have passed since last layer change (this handles continuous toolpath objects like spiral vases, segmenting them so you can look at parts of them)

This way we don't have to care about layer height or layer consistency - we just have segmented sequences of moves in 3D space. Of course this is a problem with the 2D viewer which only shows slices - but the 2D viewer can't deal with our corner cases anyway

volconst, 2020-09-17:
@kliment 
I think we can implement only 3 first and hope it will cover most of the cases giving a better experience than the current one.
The problem with spiral vases and advanced contours is that it could detect a layer for each extrusion command, which is not a big issue and will be naturally limited by the resolution of the slider - one pixel of the slider will correspond to several lines. Keyboard stepping will be slow though.
Users of layer change callbacks can very easily limit the rate of events if they need it.
At this time I am willing to implement only 3. Would you approve such PR?

kliment, 2020-09-17:
Yeah of course I'd approve! It's an improvement over the current state. I'll look into adding the others later on.

closed 2020-10-15 by volconst