kliment/Printrun #1113: pronterface.app can't be opened

Erik262, 2020-09-22:
Hello, I downloaded the version 2.0.0rc7 but I can't open it up. I always get the message "The application "pronterface" can't be opened".

I was very lucky when I had seen there was a precompiled pronterface App.

I'm running macOS Catalina 10.15.4

RyanMorash, 2020-09-27:
The app is built as a 32-bit app. macOS no longer supports 32-bit apps so pronterface will need to be rebuilt as a 64-bit app.

volconst, 2020-09-30:
@RyanMorash which file are you looking at?
I do not have a Mac, but when I download Printrun-2.0.0rc7-macos.zip from https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7 and execute file command in Ububtu, I get x86_64
```
$ file pronterface 
pronterface: Mach-O 64-bit x86_64 executable, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|PIE>
```
How do you identify that the app is 32-bit?

@Erik262 are you using the specified files?
Does this solve the problem https://github.com/kliment/Printrun/pull/1069#issuecomment-701416175  ?

a2k-hanlon, 2020-10-01:
I have the same problem with the precompiled Pronterface app on macOS (10.15.7). running `chmod +x pronterface` doesn't help, nor does a right click to open.

As an aside, when I try to package the app locally I have the same problem.

I looked at the Github Actions log for one macOS build [here](https://github.com/kliment/Printrun/runs/1182184440?check_suite_focus=true). First off, I'm pretty sure it's a 64-bit app... one pyinstaller message says `91 INFO: Platform: macOS-10.15.6-x86_64-i386-64bit`

Anyways, the main thing I wanted to mention was that there are a few warnings from pyinstaller in the Github Actions log that I think might be related to what's causing the app to crash or not open (I get these same warnings when trying to build pronterface locally too):

```
16418 WARNING: Hidden import "pkg_resources.py2_warn" not found!
16418 WARNING: Hidden import "pkg_resources.markers" not found!
```
and later
```
18908 WARNING: Ignoring /System/Library/Frameworks/IOKit.framework/IOKit imported from /Users/runner/work/Printrun/Printrun/printrun/power/osx.py - ctypes imports are only supported using bare filenames
18958 WARNING: library glu32.dll required via ctypes not found
18960 WARNING: library opengl32.dll required via ctypes not found
19349 WARNING: library libc.so.6 required via ctypes not found
19644 WARNING: library dinput8.dll required via ctypes not found
19650 WARNING: library wintab32.dll required via ctypes not found
19725 WARNING: library dsound.dll required via ctypes not found
```

When I try to run the pronterface executable (not the app, the binary file), I get the following error:
```
Traceback (most recent call last):
  File "PyInstaller/loader/rthooks/pyi_rth_pkgres.py", line 13, in <module>
  File "/Users/Andrew/code/printrun/env_printrun/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
    exec(bytecode, module.__dict__)
  File "pkg_resources/__init__.py", line 86, in <module>
ModuleNotFoundError: No module named 'pkg_resources.py2_warn'
[4420] Failed to execute script pyi_rth_pkgres
```

the fact that this mentions `pkg_resources.py2_warn`, like one pyinstaller warning, makes me think those pyinstaller messages are indicative of the problem.

As far as trying to address these warnings, I am able to mitigate the IOKit framework warning in my local build by getting rid of the filepath in the call to ctypes in `/printrun/power/osx.py`. That is, modify lines 25/26 from
```
framework = ctypes.cdll.LoadLibrary('/System/Library/Frameworks/IOKit.framework/IOKit')
```
to
```
framework = ctypes.cdll.LoadLibrary('IOKit')
```

Otherwise, I'm not immediately sure how to resolve the rest of these warnings.

kliment, 2020-10-01:
Thanks for investigating in detail. This is a setuptools-related [issue](https://github.com/pypa/setuptools/issues/1963). I'll try to fix it.

Erik262, 2020-10-01:
> @RyanMorash which file are you looking at?
> I do not have a Mac, but when I download Printrun-2.0.0rc7-macos.zip from https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7 and execute file command in Ububtu, I get x86_64
> 
> ```
> $ file pronterface 
> pronterface: Mach-O 64-bit x86_64 executable, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|PIE>
> ```
> 
> How do you identify that the app is 32-bit?
> 
> @Erik262 are you using the specified files?
> Does this solve the problem [#1069 (comment)](https://github.com/kliment/Printrun/pull/1069#issuecomment-701416175) ?

Thank you! this helped me!

PR #1117 (2021-01-29): Fix github action for building macos app
Hopefully fixes #1113.
commit 1c4a0dd2 (2021-01-29): Add hidden import to pyinstaller invocation. Change IOKit import on OSX. Hopefully fixes #1113.
closed 2020-10-01 by Erik262