kliment/Printrun #1118: Macpronterface connection to CR-10S issue

JiangXu123, 2020-10-02:
Hello, I found a weird problem about the connection between Macpronterface and CR-10S
Each time, the connection was succesfuly, but I could only successfully sent the first command to the CR-10S, such as setting the bed temperature to 60 degree centigrade, which would take effect but was followed by an error message. The other command I sent thereafter(such as setting the nozzle to 185 degree centigrade) would bring error messages and would not take effect. below is the log:
```
Connecting...
start
Printer is now online.
echo: External Reset
Marlin 1.1.9
echo: Last Updated: 2018-07-31 | Author: (none, default config)
echo:Compiled: Aug  6 2018
echo: Free Memory: 1518  PlannerBufferBytes: 1232
echo:V55 stored settings retrieved (655 bytes; crc 1179)
echo:SD card ok
Setting bed temperature to 60.000000 degrees Celsius.
Attempted to write invalid text to console, which could be due to an invalid baudrate 
Setting hotend temperature to 185.000000 degrees Celsius.
Attempted to write invalid text to console, which could be due to an invalid baudrate
Setting hotend temperature to 185.000000 degrees Celsius.
Attempted to write invalid text to console, which could be due to an invalid baudrate

```
Any suggestions?

volconst, 2020-10-03:
This could be due to non latin characters send from the printer to pronterface.
You can try connecting with pronsole, set the hotend temperature and paste the pronsole session here.

rockstorm101, 2020-12-12:
Hi @JiangXu123, were you able to stablish the source for the error? If so, could you please share the community what it was and then, if appropriate, close this issue? It did look, as @volconst suggested, like non UTF-8 characters were being sent.