kliment/Printrun #1124: stuck when call startprint multiple times

SokobanRyu, 2020-11-04:
hi printrun developer,
i found some issue on printcore 
when i call startprint on printcore the thread is stuck on _sendnext 
after investigate this i found this will be stuck after call several(many) times and always stuck
because this variable self.clear sometime set False and command is not sent

note: i run python not using cpython only plain python 3.7
the question is why this  self.clear  set to be False on start print?
thanks,

volconst, 2020-11-07:
Do you call printcore.connect(...) before startprint? It is called implicitly if you pass port and baud rate to constructor.

SokobanRyu, 2020-11-07:
Hi volconst
From my code i only call once at first time call connect via constructor

SokobanRyu, 2020-11-07:
From my several trial the clear variable is not need to set false when start print. If this values passed at the beginning "print thread" then will be stuck forever. Somehow at several time run this variable value not passed at begining "print thread" this is the mostly current happen in printcore code

volconst, 2020-11-07:
Do you use serial connection or tcp sockets?
Does the problem occur when you try to print several files, for the files after the first?
You are not trying to print concurrently several files, are you?

SokobanRyu, 2020-11-07:
Serial connection
Several list of gcode string
Im not running paralel.

volconst, 2020-11-07:
@SokobanRyu , I started a review in #1125
Maybe you can work around this by disconnecting and connecting for each printed file.

SokobanRyu, 2020-11-08:
@volconst 
Thanks for suggested but in my case i can't do disconnect and reconnect again except for serial communication timeout. The problem when i do that will be take long time.

volconst, 2020-11-08:
How long are your print jobs and how often do you invoke them?
Maybe you do not need startprint() and use send_now().

commit 96158e0f (2021-01-25): Fix send/clear race condition

M110 requires 'ok' confirm, but set it to be required before send() to
 ensure send() does not finish before
 set and the 'ok' be skipped worthless.
 Fixes #1124.
 Thanks to @SokobanRyu for reporting and coding patch.
closed 2021-01-25 by volconst