kliment/Printrun #1126: when using g29 i get these errors in terminal

tunckuyel, 2020-11-06:
when using g29 i get these errors in terminal. my printer does auto bed levelling with lcd control unit. but in pronterface does nothing and this errors in terminal. i try to do g29 with cura and there is no error. please help

File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 3, in macro
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 429, in <lambda>
  File "<string>", line 2, in macro
RuntimeError: maximum recursion depth exceeded

volconst, 2020-11-07:
The recursion seems to happen while processing a macro.
Do you have defined macros in your .pronsolerc (.ini)?
Do you invoke some macro?
Exit pronsole, rename .pronsolerc and start pronsole to test if there is still recursion.

tunckuyel, 2020-11-07:
i download the windows latest binary and there is no file named .pronsolerc ?
yes i did invoked macros but i can't find a way for cleaning. in exe version there isnt .pronsolerc

tunckuyel, 2020-11-07:
ok i did it. in macros i choose the macro and delete everything in that macro and save it. program asked do you want to delete this macro. i said yes and problem gone. it is a basic error( may be) and with your help it solved. thank you.

closed 2020-11-07 by tunckuyel