kliment/Printrun #1127: Won't work on Mac OS Catalina

tiagomaricate, 2020-11-07:
The app must be updated (64bits) to run on Catalina and newer os.

volconst, 2020-11-07:
How do you know that the app is not 64bits.
Which download are you using?
Try to set executable permission https://github.com/kliment/Printrun/pull/1069#issuecomment-701416175

rockstorm101, 2020-12-12:
Hi @tiagomaricate, did the proposed solution work? If so, would you consider closing this issue?

abcd-ca, 2021-01-10:
32 bit apps don't run on Catalina anymore @volconst . It's a change Apple made.

volconst, 2021-01-10:
@abcd-ca ,
Get 2.0 rc7 release https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7

rockstorm101, 2021-01-25:
@tiagomaricate, @abcd-ca, did you guys manage to run Printrun? Was it enough to give the launcher executable permissions?

abcd-ca, 2021-01-25:
@rockstorm101 No, I can't remember the issue but I couldn't get it working so I ended up downloading a prebuilt version from someone else on this thread. I understood the instructions and had the appropriate dev tools it was just giving me some errors that prevented me from proceeding in a reasonable amount of time. I'm sorry I don't recall the details anymore. I am also curious if the latest slicer app that was just announced contains this fix as I was reluctant to install it. The email I got was from Jan 22 and contained a heading, "PrusaSlicer 2.3 is out!"

rockstorm101, 2021-01-26:
@abcd-ca  thanks for coming back to us. Would you mind testing whether this file would work [[1]]?. Possibly using the work around explained in here [[2]] to allow execution permission. What I'm trying to ascertain here if whether the file provided works with the permissions workaround and we can therefore consider this matter closed or whether we are facing a completely different issue.

[1]: https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc7/Printrun-2.0.0rc7-macos.zip
[2]: https://github.com/kliment/Printrun/issues/1129#issuecomment-729151715

abcd-ca, 2021-02-02:
Hi @rockstorm101, I'm afraid I get "The application can't be opened", even after adding executable permissions, running `chmod +x` on the .app

a2k-hanlon, 2021-02-02:
@abcd-ca good to hear. The missing execute permission in the macOS app have been fixed, and this change is included in the newest release 2.0.0rc8. Unfortunately the security exception you have to make is still necessary because the app is not code-signed.

> The app must be updated (64bits) to run on Catalina and newer os.

@tiagomaricate The latest few releases have 64 bit binaries for macOS.

@abcd-ca and @tiagomaricate, would you both mind giving https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8 a try?

I am going to close this issue, please feel free to open a new one if you run into issues with the new release.

closed 2021-02-02 by a2k-hanlon

abcd-ca, 2021-02-02:
The security exception is something mac users need to get used to, it's pretty common now with apps.