kliment/Printrun #1129: Unable to launch the app on macOS Big Sur (kLSNoLaunchPermissionErr)

DominikPalo, 2020-11-14:
I'm trying to start the precompiled app `Printrun-2.0.0rc7-macos` on my macOS, but it always fails with a "User doesn't have permission to launch the app (managed networks)" error. I tried to use `chmod` to make the app executable (`chmod +x pronterface.app`), but it didn't help. Here is my output from the console:

```
dominik@MBP-Dominik /Applications % sudo open -a pronterface.app
The application /Applications/pronterface.app cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10826 "kLSNoLaunchPermissionErr: User doesn't have permission to launch the app (managed networks)" UserInfo={_LSFunction=_LSLaunchWithRunningboard, _LSLine=2508, NSUnderlyingError=0x7fe1b7c1d0e0 {Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x7fe1b7c0a6b0 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111" UserInfo={NSLocalizedDescription=Launchd job spawn failed with error: 111}}}}}
```

volconst, 2020-11-17:
See https://github.com/kliment/Printrun/pull/1069#issuecomment-701416175
$ chmod +x /Applications/pronterface.app/Contents/MacOS/pronterface

DominikPalo, 2020-11-17:
@volconst thank you, it helped! I didn't realize that `chmod` has to be applied to a file inside the .app, not the .app file itself. Everything is ok now :)

closed 2020-11-17 by DominikPalo

rockstorm101, 2021-01-25:
@kliment, @volconst I've seen several issues revolving around the same topic already (executable permission). Is there something we can do about that on the released .zip file itself? Otherwise, we'd better include a note about that on the README.

volconst, 2021-01-25:
If zip files include permissions, then the build procedure should run the chmod before the zip creation.
If not, maybe tar files include permissions, then distribute mac version as tar.
I do not understand the build process, do not have mac, can't fix it myself.