kliment/Printrun #1130: Pronterface on Big Sur

Iambyrdman, 2020-11-18:
Will not open on Big Sur.  Sure you know

acestu, 2020-12-05:
Hi, this is the same for me, it says that the software needs to be updated, is there any chance that this will happen ?

thanks

volconst, 2020-12-06:
I think the chance is too low.
Try https://github.com/kliment/Printrun#running-from-source

NearlyLeglessLee, 2020-12-15:
Having the same problem and running from source a bit much just to run a few test on my printer.
Hopefully there will be an update

Tolsi, 2020-12-24:
[Latest 2.0 RC](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7) works for me

Iambyrdman, 2020-12-24:
When I try to open the Mac version it tells me, "You do not have permission to open the application “pronterface”

Tolsi, 2020-12-24:
Try to execute in the terminal "chmod -R +x /.../pronterface" to the app.

a2k-hanlon, 2021-02-02:
@Tolsi the missing execute permission in the macOS app have been fixed, and this change is included in 2.0.0rc8.

@Iambyrdman Seeing as you have been able to open the app, I will close this issue.

Please give 2.0.0rc8 a try and feel free to open a new issue or reopen this one if either of you run into problems. Thanks! https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8

closed 2021-02-02 by a2k-hanlon