kliment/Printrun #1132: echo:busy: processing

nomeursie, 2020-11-27:
Hello,

When i connect my ender 3 to pronterface and do autohome , i get not information from my printer it say echo:busy: processing
I have bltouch from antclabs on my ender 3.
Can you tell what the problem is.
Best Regards Henk v Meurs

rockstorm101, 2020-12-12:
Hi @nomeursie sorry for the late response. Were you able to connect? If not, could you please tell us what operating system are you using? What version of Printrun? I'm assuming you run the pronterface app and not the console app, do any other messages come up on the messages area?

crclardy11, 2020-12-24:
I'm not the original poster, but I'm having the same issue. When I click to auto home, I get Echo:busy: Processing while the printer auto-home. If I try to send a command, I just get "Sending" but it never sends. Attached a screenshot. 
[image]

volconst, 2020-12-28:
@crclardy11 , version 1.6 although official is very old. Try rc7 or run from source for the latest updates.
Check the "Debug communications" checkbox from the menu and post the log.

rockstorm101, 2021-01-31:
Hi @crclardy11  / @nomeursie  did you try the [latest 2.0.0rc7 version][1]. Does Printrun now work now as expected?

[1]: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7

crclardy11, 2021-02-02:
Sorry for not responding. been a busy month. I had no luck with RC7, I haven't had time to go back and mess with it since trying.

Goldeneyes15, 2022-05-14:
I'm having this problem as well. After homing in pronterface the printer constantly echoes echo:busy:processing
Any commands sent don't report back

DivingDuck, 2022-05-14:
Printrun V1.6 will not run correctly with windows 10. One problem is the usb communication. This is well known. 

The actual 2.x RC versions also will not run with all localized Windows 10 versions because of the [locale wxAssertionError](https://discuss.wxpython.org/t/wxpython4-1-1-python3-8-locale-wxassertionerror/35168). This problem was solved last year with the developer version of wxPython.

You need a Printrun 2.0 RC compiled with the developer version of wxPython (actual is version 4.1.2a1.dev5310+af8cca51).

Related issues:
https://github.com/kliment/Printrun/issues/1230, https://github.com/kliment/Printrun/issues/1208, https://github.com/kliment/Printrun/issues/1170, https://github.com/kliment/Printrun/issues/1174