kliment/Printrun #1133: Problem connecting my MBP & Pronterface to my Ender 5

AllenZ-E5, 2020-12-02:
Hello

I have an early 2008 Mac Book Pro with Pronterface installed and a USB cable connect to my Ender 5?  What Port do I use?  I keep getting error messages (Errno 2)

First time attempting to use Pronterface.

Tha
[image]
[image]

nks
Allen

DivingDuck, 2020-12-02:
Maybe this help to identify your port:
https://github.com/kliment/Printrun/issues/713#issuecomment-169762414

AllenZ-E5, 2020-12-02:
Sorry, didn't help.
I am running Yosemite 10.10.5.  I don’t know what to put into the port box.

the other posts talks about returning to Yosemite.  Note how to find a port.

Thanks
Allen

rockstorm101, 2020-12-12:
Closing. Considering it sort of a duplicate of #713

closed 2020-12-12 by rockstorm101

AllenZ-E5, 2020-12-12:
Thanks
Allen