kliment/Printrun #1134: Same issue in #696, issue with CoreFoundation module when building from source on Mac

lucasgelfond, 2020-12-09:
Upgraded to OSX Catalina (10.15) which can't handle 32-bit apps. Decided to try and run Printrun from source but am getting an error associated with the CoreFoundation module;

`(printrun-venv) Lucass-Air-3:Printrun work$ python pronterface.py
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (/Users/work/Documents/Git/Printrun/printrun/__init__.py)
Traceback (most recent call last):
  File "pronterface.py", line 30, in <module>
    from printrun.pronterface import PronterApp
  File "/Users/work/Documents/Git/Printrun/printrun/pronterface.py", line 32, in <module>
    from . import pronsole
  File "/Users/work/Documents/Git/Printrun/printrun/pronsole.py", line 42, in <module>
    from .power import powerset_print_start, powerset_print_stop
  File "/Users/work/Documents/Git/Printrun/printrun/power/__init__.py", line 21, in <module>
    from .osx import inhibit_sleep_osx, deinhibit_sleep_osx
  File "/Users/work/Documents/Git/Printrun/printrun/power/osx.py", line 20, in <module>
    import CoreFoundation
ModuleNotFoundError: No module named 'CoreFoundation'
(printrun-venv) Lucass-Air-3:Printrun work$ 
`

I had thought this might be a Python package but doesn't look like it is, and saw no resolution on #696 which seems like the same issue, what should I fix?

PS—am a second-year CS student with some time on my hands. If this isn't too difficult of a programming fix, I would love to help with it/create new 64-bit Python binaries although I don't know how! Let me know/PM me though, would love to help/get some experience contributing to open source repos if I can!

kliment, 2020-12-09:
I think this is a macos-specific virtualenv issue. The OS-specific python packages in macos do not automatically get copied to a new virtual environment. You can try copying them manually with something like:

`cp -r /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/* venv/lib/python3.8/site-packages/`

adjust for your python installation's version

It would be extremely helpful if you can test running from source and building packages on macos - this is a huge problem for us at the moment as none of us have a mac available for this kind of testing.

a2k-hanlon, 2021-03-07:
@lucasgelfond It's been a few months now so I'm not sure if you're still interested in running Printrun from source, but I came across something in my command shell configuration that might be relevant to the error you ran into. When I was getting things set up on my own machine, I added

```command
export SDKROOT="/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk"
```

in my `~/.zshrc` file, which is something apparently necessary when you have the Xcode command line tools installed without Xcode as a whole (which I do). This points to the macOS SDK containing the CoreFoundation framework, among others. Perhaps you could try adding this to your `~/.zshrc` file or `~/.bashrc` instead, If you use bash rather than zsh.

Please write back if this works for you or not. Thanks!

lucasgelfond, 2021-05-10:
A bit too busy right now but let me get back to this in a week or two! Definitely want to help if I can!

a2k-hanlon, 2022-04-19:
Hi @lucasgelfond, this discussion has become quite stale for nearly a year so I'm going to close the issue now. Please open a new one if you still need help or you would like to make some contributions :)

closed 2022-04-19 by a2k-hanlon