kliment/Printrun #1136: Printerface hangs when connecting to printrbot simple.

glen-cafferty, 2020-12-17:
When I press the connect button I get the "Connecting..." message but nothing else seems to happen. When I try to press any button, then I get the "(Not Responding)" message on the Title line. Pronterface worked fine before I updated. I'm running Windows10.

rockstorm101, 2020-12-18:
On 17 December 2020 22:37:38 GMT, glen-cafferty <notifications@github.com> wrote:
>When I press the connect button I get the "Connecting..." message but
>nothing else seems to happen. When I try to press any button, then I
>get the "(Not Responding)" message on the Title line. Pronterface
>worked fine before I updated. I'm running Windows10.

Hi, what version of Printrun are you using? Are you using the latest 2.0.0~rc7 [1]?

[1]: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7

glen-cafferty, 2020-12-19:
Thanks.

My current version of Printrun is 1.6.0. I'm updating with the link you included. Will let you know if this fixed the problem.

Glen

glen-cafferty, 2020-12-19:
When i start pronterface: Microsoft Defender shows warning.

Still not working with new version. different errors.
When I connect printer, I get the following on the console:
RECV: ok T:194.9 /195.0 B:22.6 /0.0 T0:194.9 /195.0 @:47 B@:0
Printer is now online.
SENT: M105
RECV: ok T:195.0 /195.0 B:22.6 /0.0 T0:195.0 /195.0 @:45 B@:0 mailto:B@:0
(message repeats several times..)

When I try to Load a file, I select a file from the list presented and get:
Loading file: C:\Users\Glen\Desktop\CNC\CNC parts\Wire cover_export.gcode
[Errno 2] No such file or directory: 'C:\\Users\\Glen\\Desktop\\CNC\\CNC parts\\Wire cover_export.gcode'
Traceback (most recent call last):
File "printrun\pronterface.py", line 1656, in loadviz
File "printrun\gcview.py", line 337, in addfile_perlayer
File "printrun\gl\libtatlin\actors.py", line 425, in load_data
File "printrun\gcoder.py", line 250, in __len__
TypeError: object of type 'NoneType' has no len()
(File is there when I check with Notepad++. tried with several different files.)

When I press the SD  Tab, I get:
Traceback (most recent call last):
File "printrun\pronterface.py", line 629, in sdmenu
File "printrun\gcoder.py", line 250, in __len__
TypeError: object of type 'NoneType' has no len()
(No repeats)

Trying to print from Gcode, I send (receive) the following
>> M20
SENDING:M20
Begin file list
RELAYC~1.GCO
WIRECO~1.GCO
EXTRUD~1.GCO
CNCCAB~1.GCO
ENDLIN~1.GCO
PSME_E~1.GCO
End file list
>> M23 psme_e~1.gco
SENDING:M23 PSME_E~1.GCO
>>> M24
SENDING:M24
>>> M31
SENDING:M31
echo:2 min, 19 sec
Setting hotend temperature to 195 degrees Celsius.
Setting bed temperature to 0 degrees Celsius.
>>> M105
SENDING:M105
ok T:195.0 /195.0 B:22.4 /0.0 T0:195.0 /195.0 @:46 B@:0
>>> M355 S1 C1; Turn LED on Red
SENDING:M355 S1 C1
Extruding 5.000000mm of filament. (From extrude button, Did.)
>> M23 psme_e~1.gco
SENDING:M23 PSME_E~1.GCO
>>> M24
SENDING:M24
>>> M31
SENDING:M31
echo:2 min, 19 sec (but no print)
Setting hotend temperature to 195 degrees Celsius.
Setting bed temperature to 0 degrees Celsius.
>>> M105
SENDING:M105
ok T:195.0 /195.0 B:22.4 /0.0 T0:195.0 /195.0 @:46 B@:0
>>> M355 S1 C1
SENDING:M355 S1 C1; Turn LED on Red. ( LED lit)
Extruding 5.000000mm of filament.
(Pressed SD button)
Traceback (most recent call last):
File "printrun\pronterface.py", line 629, in sdmenu
File "printrun\gcoder.py", line 250, in __len__
TypeError: object of type 'NoneType' has no len()
Traceback (most recent call last):
File "printrun\pronterface.py", line 629, in sdmenu
File "printrun\gcoder.py", line 250, in __len__
TypeError: object of type 'NoneType' has no len()ressedSD
(Used printer movement controls without problem.)

I appears program (Printrun 2.0.0rc7) only has trouble with files.
The only thing I need to do are check printer status and strtt files on SD card. Any help is appreciated.

Glen

volconst, 2020-12-28:
What's the warning from Defender?
Disable Defender from Control Panel and run again.
If you insist Defender to defend you, try installing python and running from source.