kliment/Printrun #1137: IndexError: tuple index out of range and stop output messages

Tolsi, 2020-12-24:
I have pretty unstable connection (lost characters and etc) on Mac OS using Pronterface 2.0RC7 to my SKR 1.4 Turbo. I always get this error after some time after connection and some command (about 30 seconds):

```
>>> M48 P4 X160 Y100
SENDING:M48 P4 X160 Y100
M48 Z-Probe Repeatability Test
eho:buy:prcesng
echo:busy: procssing
ebs oeig
Exception in thread read thread:
Traceback (most recent call last):
  File "serial/serialposix.py", line 500, in read
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 333, in _readline
  File "printrun/printcore.py", line 329, in _readline_nb
  File "serial/serialposix.py", line 509, in read
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun/printcore.py", line 420, in _listen
  File "printrun/printcore.py", line 354, in _readline
IndexError: tuple index out of range
```

After that, no message comes from the printer until reconnection.
I tried to decrease the printer connection speed (250000 to 115200) but nothing changed.

volconst, 2020-12-28:
" multiple access on port?"
Is it possible that another program - pronterface or slicer is connected to the same serial port?
Check in the task manager if you have several pronterface processes.

rockstorm101, 2021-01-31:
Closing due to lack of feedback. Please feel free to reopen if needed.

closed 2021-01-31 by rockstorm101

Tolsi, 2021-02-11:
Sorry for the long delay.

Hmm, I think I have Cura running a lot when using Pronterface. Maybe I did
not notice and it tries to somehow use it with a printer without asking.  I
disabled USB plugin in it and not it shouldn't do this.

BTW I increased a buffer for the serial port in marlin (BUFSIZE in adv
config) and reduced the speed to 57600 and now the characters do not
disappear and the connection has become more stable.  most likely this is
not a Pronterface bug.

KempfCreative, 2021-12-15:
I had this same issue and it was resolved by closing Cura.