kliment/Printrun #1138: Pb with Mac OS Big version 11.1

JNRAY, 2020-12-31:
Hello,

The program does not work with Mc OS Big version 11.1.

Do you plan to update it.

Regards.

volconst, 2021-01-02:
See #1129

JNRAY, 2021-01-03:
> See #1129

Hello,
I have tried to create Python virtual environment as describe in the Printrun readme file but it did not work.
Could you provide me the executable file ?

Thanks in advance.

Jean Noël RAYNAUD

volconst, 2021-01-04:
https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc7/Printrun-2.0.0rc7-macos.zip

Rmndbrvn, 2021-01-10:
> https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc7/Printrun-2.0.0rc7-macos.zip

You do not have permission to open the application “pronterface”.
Contact your computer or network administrator for assistance.

rockstorm101, 2021-01-25:
Hi @JNRAY, did you resolve your issues? If so, could you please share what it was that you were missing with the community? Also, could we close the issue then?

a2k-hanlon, 2021-02-02:
@Rmndbrvn Thanks for sharing that, the issue with missing execute permissions in the macOS app has now been resolved in 2.0.0rc8. Please see https://github.com/kliment/Printrun/releases/latest

a2k-hanlon, 2021-02-25:
Closing due to lack of feedback since Printrun has been reported working on macOS 11 at https://github.com/kliment/Printrun/issues/1016#issuecomment-773490020, and as above, the execute permissions problem has been fixed.

closed 2021-02-25 by a2k-hanlon