kliment/Printrun #1139: Pronterface won't connect to my CR10S Pro

ducttapealien, 2020-12-31:
When I connect the printer to my computer I can click connect but it never recognizes the printer as active. It reads back some settings like Marlin version (2.0.6_DW7), probe offset and SD card ok. But if I do a m503 it returns "(error) printer is not active". 
I have tried different ports numbers in device manager, I have installed the CH341 driver on my pc, I am set to 250000 Baud. 
I am desperate for any help getting a stable connection because I need to modify some of my machine settings inorder to correct my probe x/y offset.

volconst, 2021-01-02:
If it reads marlin version, then port is correct.
Are you using https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7 ?
Go to menu, and check Debug Communications checkbox. Then reproduce and paste the log.

ducttapealien, 2021-01-02:
Connecting...
echo:start
 External Reset
Marlin 2.0.6_DW7
echo: Last Updated: 2020-08-10 | Author: TinyMachines3D
echo:Compiled: Sep 25 2020
echo: Free Memory: 1438  PlannerBufferBytes: 736
==Dwin Init Complete==
//action:notification TM3D 10S Pro Ready.
==onLoadSettings==
echo:V81 stored settings retrieved (757 bytes; crc 40200)
==onConfigurationStoreRead==
 init zprobe_zoffset = -1.70
//action:prompt_end
echo:SD card ok
  startprogress
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
SENT: M105
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed (WindowsError(5, 'Access is denied.'))
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
SENT: M105
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed (WindowsError(5, 'Access is denied.'))
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
SENT: M105
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed (WindowsError(5, 'Access is denied.'))
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
SENT: M105
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed (WindowsError(5, 'Access is denied.'))
[ERROR] Aborting connection attempt after 4 failed writes.
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed

volconst, 2021-01-03:
This is probably caused by https://github.com/MarlinFirmware/Marlin/issues/20305
Try to send this command:
!self.p.greetings.append('echo:start')
in pronterface before connecting. If it works you can define a custom button and press it after starting pronterface.

ducttapealien, 2021-01-18:
I apologize for not getting back to this sooner but life interfered, 
I sent that command like you suggested and it worked like a charm. I now have a green custom button at the bottom of the screen called Pre Connect and i am able to interact with my printer through Pronterface when ever i want. 
Thank you very much.

volconst, 2021-01-18:
Great, thanks !

closed 2021-01-18 by volconst