kliment/Printrun #1140: Projector fails displaying layer

uli99, 2021-01-02:
Hi,

OS: Linux Mint 20 Cinnamon
Pronterface 2.0.0rc7

Just tried to use Pronterfaces Projector tool to display/ project a SVG slice created with Slic3r (Slice to SVG). The file displays fine with:
http://garyhodgson.github.io/slic3rsvgviewer/

I clicked on Controls-->Load and loaded the SVG-Slice-File with no error and displays correct infos in the Info section:
File: gear_for_wormgear_lychee.svg
Total Layers: 833
Current Layer: 0
Estimated Time: 01:09:25

But when I click on the "Present" button or try to select the "1st Layer" checkbox I get the error message below:

Layer thickness detected: 2.999999999999802e-08 mm
833 layers found, total height 2.498999999999835e-05 mm
/home/uli/Documents/3D-Printer_LCD/Printrun/printrun/projectlayer.py:121: wxPyDeprecationWarning: Call to deprecated item BitmapFromImage. Use :class:`wx.Bitmap` instead
  dc.DrawBitmap(wx.BitmapFromImage(image.Scale(image.Width * self.scale, image.Height * self.scale)), self.offset[0], -self.offset[1], True)
Traceback (most recent call last):
  File "/home/uli/Documents/3D-Printer_LCD/Printrun/printrun/projectlayer.py", line 614, in show_first_layer
    self.present_first_layer(event)
  File "/home/uli/Documents/3D-Printer_LCD/Printrun/printrun/projectlayer.py", line 676, in present_first_layer
    self.display_frame.draw_layer(copy.deepcopy(self.layers[0][0]))
  File "/home/uli/Documents/3D-Printer_LCD/Printrun/printrun/projectlayer.py", line 105, in draw_layer
    stream = io.StringIO(PNGSurface.convert(dpi = self.dpi, bytestring = xml.etree.ElementTree.tostring(image)))
TypeError: initial_value must be str or None, not bytes

I'm a first time user of Pronterface so it might be a stupid mistake on my side...

Uli

volconst, 2021-01-02:
Are you starting pronterface with python3?
Pass only the -V switch to the interpreter you are using.
Not sure if projector code is working.

uli99, 2021-01-02:
I started it with 
Python 2.7.18
I just tried 
Python 3.8.5
but with the same outcome.

I think the problem is that projectlayer.py tries to load the slic3r SVG's namespace from
http://slic3r.org/namespaces/slic3r
in line 501and fails to do so because the url is no longer valid (page not found).

I tried to find a working url but couldn't find anything usful (yet).

volconst, 2021-01-08:
@uli99 You can checkout this branch https://github.com/volconst/Printrun/tree/projector-api-sync and test the fix.
The branch fixes the exception, tooltips and layout, but I have no idea whether you can actually print an object with Projector.

uli99, 2021-01-08:
Seems to work (despite the not working namspace url, not sure why?). Maybe I don't understand the code that well...
I had to replace time.clock() with time.time() to get it to run.

Thanks!

volconst, 2021-01-08:
I think namespaces are URIs, not URLs. Just names, not an existing resource that is retrieved.
Does it throw exception for time.time().
Can you print an object now?

uli99, 2021-01-10:
Still have to modify the script for my displays resolution (2K display with 2560x1600px) and use a Raspberry 4 with two displays to  test...

volconst, 2021-01-10:
Does'nt a usable projector host software come with your printer?
Aren't "X", "Y", or "Fullscreen" dialog controls enough to adapt to different screen sizes?
Do you find the log using time()/clock() useful? I do not know what it's purpose is, probably to track whether rendering load messes up slide timing, but you have to subtract the times yourself.

PR #1148 (2021-01-25): Projector fix, race condition fix
Fix #1140 #1124
commit cf00197f (2021-01-25): Fix #1140 - Projector using obsolete API

Rendering did not work, probably since python3.
Default first layer timeout of -1 was ignored
closed 2021-01-25 by volconst