kliment/Printrun #1142: Console infinite scroll

peterpasha, 2021-01-13:
I opened the options menu in Pronterface to change the bed size parameters from 0.00 to 217.26 and clicked okay. The console started to scroll infinitely. I can not stop it or interrupt it. I can still connect to the printer and perform simple commands like G28. Anything that would need a display like M501 won't work. I've changed the all the baud settings from 2500000 to 2400 without effect. I've reflashed the firmware on my printer without effect. I can still open the options menu, but any changes I make don't remain once I close the menu. I need to use Ctrl-Alt-Delete and the Task Manager to close Pronterface. I'm running Printrun-win-18Nov2017 on a Windows 10 computer
5 second video of the infinite scroll: [](https://www.facebook.com/890335524/videos/10164879535155525/)

DivingDuck, 2021-01-13:
Printrun 1.6 and actual Windows 10 isn't a good combination these days. Pls. check out the newer version from here: [printrun 2.0.0RC7](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7)

peterpasha, 2021-01-14:
definitely a typo. I've reflashed marlin on the board and changed all the different baud settings with no effect. But I'm happy to see a new version, I'll give it a try.

peterpasha, 2021-01-14:
When I opened the new version of Pronterface, this is what was displayed in the console:

Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
RuntimeError: wrapped C/C++ object of type XYButtonsMini has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
RuntimeError: wrapped C/C++ object of type TempGauge has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
RuntimeError: wrapped C/C++ object of type TempGauge has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 2311, in Notify
  File "wx\core.py", line 3508, in Notify
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
RuntimeError: wrapped C/C++ object of type Gviz has been deleted

I'm assuming that this is what was looping on infinite scroll, if this helps.

volconst, 2021-01-14:
I have seen this after editing settings. I think it is harmless. Should not show on a fresh start

DivingDuck, 2021-01-16:
@peterpasha, has the actual release candidate solve your problem?

peterpasha, 2021-01-16:
Yes, the problem is solved. Thanks for the help!

closed 2021-01-16 by peterpasha