kliment/Printrun #1145: error: invalid command 'bdist_wheel'

bryanpedini, 2021-01-22:
OS: Debian 10 buster (all updated)
Python --version: Python 3.7.3
Encountered error:
```
Building wheels for collected packages: wxPython, cairocffi, psutil, dbus-python                                                                                                                                                             
Running setup.py bdist_wheel for wxPython ... error                                                                                                                                                                                        
Complete output from command /home/bryanpedini/bin/Printrun/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ywitft47/wxPython/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-lw37icb0 --python-tag cp37:                                                                                                                        
usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]                                                                                                                                                                            
or: -c --help [cmd1 cmd2 ...]                                                                                                                                                                                                           
or: -c --help-commands                                                                                                                                                                                                                  
or: -c cmd --help                                                                                                                                                                                                                       
error: invalid command 'bdist_wheel'                                                                                                                                                                                                       
----------------------------------------                                                                                                                                                                                                   
Failed building wheel for wxPython
Running setup.py clean for wxPython
Running setup.py bdist_wheel for cairocffi ... error
Complete output from command /home/bryanpedini/bin/Printrun/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ywitft47/cairocffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-cl42ogts --python-tag cp37:
usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
or: -c --help [cmd1 cmd2 ...]
or: -c --help-commands
or: -c cmd --help
error: invalid command 'bdist_wheel'
[... (all the same) ...]
```
Full log: https://gist.github.com/bryanpedini/b96c74955c9c9e882b247064e5c8378e

hroncok, 2021-01-22:
Try `pip install wheel`, but I recommend to fetch wxpython wheels from https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ as documented in the readme.

bryanpedini, 2021-01-22:
Thank you for the suggestion, however (also from the complete log I attached), I already performed said action, directly from extras.wxpython.org (following official README, Linux section, "run from source").

rockstorm101, 2021-01-23:
Hi @bryanpedini, new Debian 11 Bullseye (which will be released in a matter of months) will ship the latest Printrun. I won't be backporting it into Buster because the new release is almost out already. I'm sorry for this inconvenience.

bryanpedini, 2021-01-23:
So just 'cause a new OS has "hard freeze" planned 1 month and a half from now, with full release expected around another two months from said time, that means that we'll not work on solving issues anymore?
If that was the approach with every open source project Linux might have been dead 5 days after it was first released.

Anyway, I really hope there will be no such errors in the next Debian release; we as 3D printing community **need** an interface with our printers, and Printrun has become the defacto standard (not mentioning it's one of the most advanced and feature-rich).

bryanpedini, 2021-01-24:
> Not even taking a look if it was actually a 5 minute fix?

It wasn't 5 minutes, more like two hours of trial and error with various Dockerfile(s) to even try to use your stupid `testing` (bullseye) version of Debian, but after using the error messages from the Docker containers, I found where the problem lied.

The issue was that I didn't have some package installed via APT and I needed to do `python -m pip install wheel && python -m pip install python-config` to fix the error mentioned in the title; plus, the instructions on the `README.md` are wrong, since there is already `wxPython==4.1.0` in the `requirements.txt` there is no need to download it separately from a website in which you have to discover which OS and which file to download (which will fail).

Yes, afterall, if I was any little more experienced with Python in general and a little bit more acute in reading the error logs, it would have been a 5 minutes fix. Something you should even be able to spot in 5 seconds or less, since you develop this program yourself, right?

I gave you compliments on your program, I uploaded an entire log history of what happened, I mentioned the error message, I've done my Google searching before coming here and asking for help, and I receive a "go F yourself, we don't have time since next release is here, now let's downvote you with sad emojis so you feel bad 'cause you asked help to us superior species" response...?

Well, at least now it works, and not thanks to you. Downvote me with sad emojis how you want, still, 5 minutes fix if I was any little bit more expert, which you should be!

closed 2021-01-24 by bryanpedini

hroncok, 2021-01-24:
@bryanpedini Please keep your insults to yourself and mind your attitude. Nobody here is obligated to fix your problems.

kliment, 2021-01-24:
@bryanpedini This is not an acceptable way to talk to people who are trying to help you. Please go away.