kliment/Printrun #1155: Layer count differ from slicer's

volconst, 2021-01-31:
Hi @volconst and @kliment 
can you give me a favor and load the attached gcode file in your latest build and then open the Gcode view?
The file should have a z high of 15mm with 75 layers. Pronterface tells me 77 Layers and Z=15.500mm. In addition the first layer says Layer 1 - Z = 15mm
[Pronterface_test.zip](https://github.com/kliment/Printrun/files/5896504/Pronterface_test.zip)

_Originally posted by @DivingDuck in https://github.com/kliment/Printrun/issues/1069#issuecomment-770091308_

volconst, 2021-01-31:
One extra layer added by printrun is GCode.append_layer at the end of gcode parsing.
append_layer is referenced in GCode.append(self, command, store = True)
but append is called only once with store = False.
I think we can remove GCode.append_layer.
@kliment do you see any use of GCode.append_layer ?

kliment, 2021-01-31:
I don't, but it's been so long since I touched any of that that I'm not certain about it.

volconst, 2021-02-01:
@DivingDuck , can you test layer counting in this branch https://github.com/volconst/Printrun/tree/layer-counts ?

volconst, 2021-02-01:
@kliment , append_layer is probably meant to allow injecting of gcode at the end of the file.
Injecting adds to the beginning of layer and adding an empty layer allows to add at the end of the file.
But I could not select the append layer with the slider.
I dropped the idea to remove append_layer as it does not affect counting with the currently proposed changes.

DivingDuck, 2021-02-02:
@volconst, the layer count of this version is correct and I see the behavior you mentioned as well. 
After injecting additional code and then saving the gcode in a new file I can see two injections, one on the position I inject the code and an additional empty line at the end of the file.

volconst, 2021-02-02:
Thank you, @DivingDuck

DivingDuck, 2021-02-02:
You're welcome, and thank you too, @volconst   :)

PR #1158 (2021-02-02): Refine layer counting
Fixes #1155
commit 1618ae51 (2021-02-02): Refine layer counting. Fix #1155

Do not count as a layer a height at which we have (+ or -) extrusion,
but no x,y movement. This excludes retraction and  priming.
Apply the same condition for end layer.
closed 2021-02-02 by kliment