kliment/Printrun #1156: Zoom in 3D view behaves wrong on macOS in high resolution

a2k-hanlon, 2021-01-31:
So after putting #1153 together and comparing the app before/after the changes, I've come across a minor side-effect of rendering the app at the full resolution offered by a "retina" screen on a Mac. When the app is rendered at this full resolution, the 3D view does not zoom in/out on the cursor position. Instead, the zoom centres on some x,y position that seems scaled towards 0,0 from where it should be.

For example, if I line the cursor up to the middle of the size 200x200 grid (ie. to point 100,100) and scroll to zoom in or out, the view zooms on some other point, around 95,95 or so.

This happens when I run the packaged app (from commit 862b7e1, ie. with the changes of #1153) as well as running from source with or without the #1153 changes (on a mac with a retina screen).

I can disable the full resolution capability of the packaged app (without rebuilding). What's interesting is after doing so, I no longer see the incorrect zoom behaviour (but obviously the overall resolution of the interface is reduced).

I'm not sure if this is an issue that can be fixed within Printrun or if it is more a Pyglet issue. My guess would be it's Pyglet, but I might be wrong. This isn't a huge issue; the zoom still mostly functions but it's just more of an nuisance than anything.

kliment, 2021-02-01:
I suspect it's an issue on our end - somewhere in that code we are using screen pixels instead of device-independent pixels. See https://docs.wxpython.org/high_dpi_overview.html for details.

a2k-hanlon, 2021-02-01:
Oh that is interesting, thanks for sharing that info. I think I stand corrected then, and I have a feeling I've opened up a can of worms here... but maybe fixing this means no more than applying a scaling factor like [GetContentScaleFactor](https://docs.wxpython.org/wx.Window.html#wx.Window.GetContentScaleFactor) in a few key places.

I also noticed that this issue affects the move and orbit operations in the view as well as the zoom; they're all out by a factor of 2.

kliment, 2021-02-01:
wx correctly handles scaling on macos for the parts of the UI that it draws itself, but those changes don't make it across to the gl view correctly. Therefore mouse events going into the gl window are incorrectly scaled. I've seen this on several applications that use mixed gl and non-gl user interfaces.

So the problem is finding those key places. But now that the can of worms is open, we should cook them before they spoil :)
I think the relevant places are in printrun/gl/panel.py in the functions starting from line 337. I am not sure whether we need to apply the scale factor to the spots that use self.height/self.width or to GetPosition. I also don't know whether we need to scale the result of GetClientSize. Do you feel like playing around with this?

a2k-hanlon, 2021-02-01:
Ah ok I follow. Sure, with the helpful insight you've shared, I can try playing around with it.

PR #1157 (2021-02-02): Fix poor 3D view interaction behaviour on high-DPI screens
Fixes #1156. In short,

[image]

I had to scale all the return values of `event.GetPosition()` and `self.GetClientSize()` by `self.GetContentScaleFactor()`. That's all.

There's only one edge case where the behaviour is still sub-optimal and I don't know how to fix it. It's kind of a niche issue. I have my computer (a mac with a Retina screen) connected to an external display with a lower DPI. When I move the app from one screen to the other, the `width` and `height` attributes of the `wxGLPanel` instance aren't updated for the differe [... cut]
closed 2021-02-02 by kliment