kliment/Printrun #1159: Pronterface is detected on VirusTotal

qwewer0, 2021-02-02:
### Description
Not sure if it is anything to be concerned or just a false alarm, but [2.0.0rc7](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc7) and [2.0.0rc8](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8) is detected on VirusTotal.

VirusTotal detection:
- [x64 2.0.0rc7](https://www.virustotal.com/gui/file/00fd0cf7f68f2d60229f63c4e750020a81d20c1c7b3f1950c745ab82ed354bac/detection)
- [x86 2.0.0rc8](https://www.virustotal.com/gui/file/ede19b9e1c3a88d70d7be54daf727be8b9be862a985fd0f7fc8b83c9f27bf8b9/detection)
- [x64 2.0.0rc8](https://www.virustotal.com/gui/file/3d6a2ed084e6d222c9e7e7613bfd19b3930962634de36bf87b8f78cee91e7f7a/detection)

DivingDuck, 2021-02-02:
It's a unknown executable for most AV's. This will change over time as soon as more people will use the new release.

qwewer0, 2021-02-02:
@DivingDuck Thanks for the info.

closed 2021-02-02 by qwewer0