kliment/Printrun #1160: Misc UI glitches

VanessaE, 2021-02-02:
Using commit 5d0ec6d25dd67fc51afccf028d4e77606ec81900 under Debian "Bullseye", with XFCE 4.16.

- [ ] Changing UI modes throws errors

    If you start the program out in Compact + Mini UI layout mode, and switch to Standard + Standard layout mode, the program hangs-up for a few seconds, then it throws a bunch of errors to the console (log below).  This will cause a couple of brief pauses if there's a print ongoing, but visually everything looks normal once it comes back, and the print continues without issue.

- [ ] Changing UI modes breaks window sizing

    After switching UI modes, something is left in the window's properties such that XCFE can no longer automatically size the window to 1/4 of the screen size when you drag it into one corner.  However, I can still manually resize it to fit.

- [ ] Changing UI modes breaks console unhide button

    Start out in Compact + Mini UI mode.  Switch to Standard + Standard UI layout, hide the console on the right, hide the controls/monitors on the left, then UN-hide the controls/monitors.  The console's unhide button gets pushed out of the window.  You now have to make the window super-wide to reveal the missing button, and you can no longer shrink the window back down to a reasonable size, or it'll just hide the console again.

(in the above, by "start out", I mean set whatever mode is called for, then close the program and relaunch it)

<details><summary>The above-mentioned log paste...</summary>
```
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 2311, in Notify
    self.notify()
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3508, in Notify
    self.result = self.callable(*self.args, **self.kwargs)
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 2311, in Notify
    self.notify()
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3508, in Notify
    self.result = self.callable(*self.args, **self.kwargs)
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
RuntimeError: wrapped C/C++ object of type XYButtonsMini has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
RuntimeError: wrapped C/C++ object of type TempGauge has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
RuntimeError: wrapped C/C++ object of type TempGauge has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/vanessa/RepRap/Printrun/printrun/gui/graph.py", line 391, in StartPlotting
    self.Refresh()
RuntimeError: wrapped C/C++ object of type Graph has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 2311, in Notify
    self.notify()
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3508, in Notify
    self.result = self.callable(*self.args, **self.kwargs)
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 2311, in Notify
    self.notify()
  File "/home/vanessa/RepRap/Printrun/venv/lib/python3.9/site-packages/wx/core.py", line 3508, in Notify
    self.result = self.callable(*self.args, **self.kwargs)
  File "/home/vanessa/RepRap/Printrun/printrun/gl/panel.py", line 139, in Refresh
    return super().Refresh(eraseback)
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
```
</details>

<details><summary>IRC chat log about these glitches</summary>

```
[02-02 17:14] <VanessaE> I switched the interface from Compact + Mini to Standard + Standard.  The program hung up for a few seconds, then dumped this error to the console:  [pastebin link containing the above error dump]   The ongoing print seems to be unaffected.
[02-02 17:15] <VanessaE> Kliment: ^  also, if I hide the right panel (console) and then hide the left panel (controls + monitors), and then UN-hide the left panel, the right panel's unhide icon gets pushed off-screen
[02-02 17:15] <VanessaE> I have to make the window super-wide to reveal the hidden icon/widget.
[02-02 17:16] <VanessaE> and I can no longer shrink the window back down or the console gets hidden again
[02-02 17:17] <VanessaE> oh, correction to the above-- when the UI hangs, there IS a brief pause in the print
[02-02 17:18] <Kliment> VanessaE: please report all those on github, I can't keep up
[02-02 17:18] <VanessaE> also changing the UI back and forth like that breaks something such that XFCE can no longer auto-tile the window to 1/4 my screen size, though I can manually resize it
```
</details>

rockstorm101, 2021-02-03:
Thanks @VanessaE. I'm able to reproduce 1 and 3 on my machine running Debian Sid with MATE 1.24.1. Same behaviour using Printrun 2.0.0rc7.
- It is not a regression of 2.0.0rc8
- It is not specific to the XFCE desktop environment

DivingDuck, 2021-02-03:
@rockstorm101, I can confirm the same behavior for Windows for too.

a2k-hanlon, 2021-02-04:
On macOS, I can only reproduce number 1. I tried a few times to reproduce 3, hope I'm not doing something differently.

I guess I could also mention a different issue related to changing the UI layout here, and that is the "interface mode" selector in the options window under the user interface tab lets you type in it. It doesn't seem to break anything if you type in something other than the drop down choices "Standard" or "Compact" but that style of setting probably wasn't intended.
[image]

I think this would be a quick fix, just change
https://github.com/kliment/Printrun/blob/5d0ec6d25dd67fc51afccf028d4e77606ec81900/printrun/pronterface.py#L963-L964
to
```python
self.settings._add(ComboSetting("uimode", _("Standard"), (_("Standard"), _("Compact")) , _("Interface mode"), _("Standard interface is a one-page, three columns layout with controls/visualization/log\nCompact mode is a one-page, two columns layout with controls + log/visualization"), "UI"), self.reload_ui)
#self.settings._add(ComboSetting("uimode", _("Standard"), (_("Standard"), _("Compact"), _("Tabbed"), _("Tabbed with platers")), _("Interface mode"), _("Standard interface is a one-page, three columns layout with controls/visualization/log\nCompact mode is a one-page, two columns layout with controls + log/visualization"), "UI"), self.reload_ui)
```

volconst, 2021-02-04:
@a2k-hanlon , I think this is a hack to support the commented Tab uimodes if they are persisted in settings. Thus they are not advertised, but still supported.

a2k-hanlon, 2021-02-04:
Ohhh that didn't occur to me; it shows that I'm new here 😄. Alright disregard that about the option selector then. Thanks @volconst

volconst, 2021-02-04:
You are welcome :)
You can look at the 'Blame' view, to dig into history of the lines of interest.