kliment/Printrun #1164: What's it all about

jimblezz, 2021-02-05:
I am new to 3D printing, this software came with the slicer. I have read the stuff here but I have no idea what any of it is about. Is there an English Language introduction I am 75 years old and not into this computer stuff. But if Prusa thought I should have this programme/app/whatever then please tell me how it will help. 
I have Prusa SL1 which is great, but struggling with the Prusa slicer to get files that will print successfully! The provided sliced examples are great - but I want to design/print my own stuff.
Any cluues or pointers much appreciated!
Jim

VinnyCordeiro, 2021-02-05:
Greetings! Printrun is a suite of programs with the purpose of sending g-code to the printer. G-code is the file format that slicers generates for 3d printers, as you may already know. The main advantage of Pronterface (the graphical user interface program) and Pronsole (the command line program) is to manually send g-code commands directly to the printer, there are some situations when that is useful.

I don't know how PrusaSlicer performs for resin printers as I don't own one myself, this slicer was originally made for FDM printers (the ones that melts plastic filaments) and was adapted later, but maybe you could try other slicers made specifically for resin printers. The most popular is ChiTuBox, but recently LycheeSlicer started to gain traction.

kliment, 2021-02-05:
PrusaSlicer works wonderfully for the SL1, but you don't need Printrun for that - like @VinnyCordeiro said it's meant for the other kind of printer they sell. They have the same software package for both.

kliment, 2021-02-05:
For designing your own models, you want to look into applications like blender, openscad, or tinkercad. You need to export your files in STL format for them to work with your printer. If you want to make your own models you will need to learn a little bit about 3d modeling/design. For more artistic models, Blender is probably the thing to use - it's an artist's tool, and quite complex and powerful, but there are excellent online resources for it. For more technical designs you can look at openscad (if you're not afraid of things that look a bit like programming, but it's actually quite easy to learn) or tinkercad (it's a browser based 3d modeling tool that is more visually oriented). Have fun!

jimblezz, 2021-02-07:
Many thanks to both of you, I think I am beginning to understand - it takes a long while to learn these new concepts and even longer for them to 'stick'!

jimblezz, 2021-02-07:
On 3D design, I am also learning Fusion 360 - and that is also complex - at least to me. Still I like a challenge.

DivingDuck, 2021-02-07:
Hope you have already found Lars Christensen for Fusion 360. He makes really good tutorials for Fusion360.

kliment, 2021-02-07:
I've not used Fusion (not really comfortable with the way they do business) but if it works for you, it works for you. If you don't mind, I'll close this issue as it's not really anything to do with Printrun.

closed 2021-02-07 by kliment