kliment/Printrun #1165: No proper pip wxpython version for gentoo

TechieAndroid, 2021-02-07:
I'm a little stumped here:
`(venv) $ python -m pip install https://extras.wxpython.org/wxPython4/extras/linux/gtk3/fedora-27/wxPython-4.0.1-cp36-cp36m-linux_x86_64.whl  # replace the link with yours`
Is there another way to do this for a distro like gentoo?

kliment, 2021-02-07:
I don't know if there's a wxpython wheel for gentoo. You can edit requirements.txt and put in the version you have installed and skip this step, and the next pip install command should find your system wxpython.

kliment, 2021-02-07:
To check whether your system wxpython is installed correctly, open a python terminal and type in `import wx` - that's what it tries to do just before emitting that error.

TechieAndroid, 2021-02-08:
Okay so I tried what you said and I thought it imported it fine (since there was no output to that command) but when I went to run the pip install requirements command (after adjusting the wxpython version to my system's version) it failed, saying I needed to install wheel, after doing so I tried again and now it fails with:

`Building wheels for collected packages: wxPython, cairocffi, dbus-python
  Building wheel for wxPython (setup.py) ... error
  ERROR: Command errored out with exit status 1:`
  
 I only copied the beginning of the error because it was extremely long.
 
I really wanted it to work in gentoo but this build process is just out of my norm and I don't know enough python to start trying to fix something like this.
For now, I'll just use pronterface in my Arch partition.

Thank you for trying to help me though.

closed 2021-02-08 by TechieAndroid

CocolinoFan, 2023-08-03:
@kliment Thank you for info. Can you be more precise on how to edit requirements.txt and put in the version I have installed?
I have version 4.2 found here https://packages.gentoo.org/packages/dev-python/wxpython

DivingDuck, 2023-08-03:
Hi,
for the actual Pronterface version (2.0.1) you need wxPython version >= 4.2 for Python 3.10. You don't need to update requirements.txt if you install the appropriate wxPython version before running the requirements.txt as it is then already version 4.2.