kliment/Printrun #1166: Pronterface does not recongize that printer is reset and continues to print

pedjas, 2021-02-08:
When it happens printing job went wrong way there is no simple way to stop printing from Pronterface. I usually end up hitting reset on printer.

Surprisingly Pronterface does not recognize printer is reset but keeps sending print commands.

DivingDuck, 2021-02-09:
How should Pronterface knew what your printer or yourself, is doing except from what it is telling to do? This is like to pull the rug out from under your feet. Simply don't do if you want use Pronterface as host. Manage your print with Pronterface as it is controlling your printer.

kliment, 2021-02-09:
Actually, we can probably detect this - doesn't the firmware send "start" on a line on its own after a reset? We can at least pause when that happens mid-print.

DivingDuck, 2021-02-09:
Hi @Kliment,
If I remember correct, there was one who had problems too in the past, but I'm not sure.

I did a quick test on my printer as I'm just sit beside it:
Print multiple cycles and reset during the print

```
SENDING:test circle 120 10 20000
G91 G0 X-120.000000 F20000.000000 G90
G2 I120.000000 J0 F20000.000000
G2 I120.000000 J0 F20000.000000
G2 I120.000000 J0 F20000.000000
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Das Gerät erkennt den Befehl nicht.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 926, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range

Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "printrun\printcore.py", line 744, in _send
  File "serial\serialwin32.py", line 317, in write
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'Das Gerät erkennt den Befehl nicht.', None, 22))
```

kliment, 2021-02-09:
@DivingDuck yeah this is what happens when the printer microcontroller has native usb - what @pedjas is reporting happens when it's a separate chip attached with a hardware uart.

DivingDuck, 2021-02-09:
Ah, ok that's different, Hadn't read that from the post.

pedjas, 2021-02-10:
> 
> How should Pronterface knew what your printer or yourself, is doing except from what it is telling to do? This is like 
>to pull the rug out from under your feet. Simply don't do if you want use Pronterface as host. Manage your print with >Pronterface as it is controlling your printer.

As I said, I did not manage to stopping printing from Pronterface so I had to go to reset option.

As printer sends various info during initialization after reset I suppose Pronterface could use that info, or at least recognize it does not get confirmations for sent commands as expected and react to that. It just continues to print regardless what happens with printer.

I guess there are other situations other than user reset that printer can fail and report error that should indicate Pronterface to stop printing.

> Edit: Forgot to mention that you can stop a print in different ways. one is the Off button. 

That did nothing on my Pruša printer. But I will try again.

> A other way is to add a custom button with a specific g-code/m-code your printer know for an e.g. emergency stop.

Did not know that is possible. I am more end user than developer. I will try to find how to do it.

kliment, 2021-02-10:
@pedjas I think the core problem here is that your printer is configured to ignore reset commands from firmware. The off button stops sending commands and turns off heaters and motors. However, this does not take effect immediately - the printer has a buffer of commands and will not stop until it has finished them all. There might be a special command to stop immediately, but I'm not sure what that is.

Could you perhaps enable communication debugging in pronterface and then record what comes back from the printer when it gets reset mid-print? It doesn''t have to be an actual print, you can just make a gcode file that moves one axis back and forth many times to test this. Then we can see what the printer says and possibly detect the reset and stop the print.