kliment/Printrun #1167: How to connect with the Linux Verstion?

DadOfThree, 2021-02-08:
Hi Folks,

I'm a Microsoft user migrating to Linux.  I have Pronterface installed on Ubuntu.  I can not figure out how to get this to connect to my Ender 3 Pro.  In windows, I would see what port the Ender is connected to in Device Manager, then I'd make sure I select that port in Pronterface.  With the Linux version, I'm not sure what to do? 

Some Guidance would be appreciated.

volconst, 2021-02-08:
Ports in linux are /dev/* files. The ports dropdown should be populated with found ports.
If not, you may need a driver for the serial chip of the printer or just identify the /dev/ file if it has driver but the filename pattern is not recognized by pronterface.
Try the lsmod, lsusb, dmesg commands for more info.

DadOfThree, 2021-02-09:
I may have jumped into Ubuntu Linux to fast and this is problematic because I really need to connect to my printer.  The commands you mentioned are unfamiliar to me.  

The port section of Pronterface shows, "/dev/ttyACM0".
When I try to connect, I get the message below:

"Connecting...
Could not connect to /dev/ttyACM0 at baudrate 9600:
Serial error: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'"

DivingDuck, 2021-02-09:
Baud rate need to match with what ever is set in your printer firmware.
9600 seems a quite low with today's hardware. Try 115200

dlarue, 2021-02-09:
Linux is multi-user so you probably need to add your user ID to the Dialout group so that you can access that device your printer is using. To list what the permissions are on the ttyACM0 device use this command for a long listing( the -l defines a long listing):
ls -l /dev/ttyACM0

To add your userID to the dialout group use this command(enter your user ID where it says YOURUSERID):
sudo adduser YOURUSERID dialout

try to connect and report back.

volconst, 2021-02-09:
> sudo adduser YOURUSERID dialout

You will probably need to logout and login into your desktop environment after this.

steveb85, 2021-03-16:
Hey guys, did this get sorted? I have the same problem here, dialout group didn’t help me at all same with baud rate.

volconst, 2021-03-16:
@steveb85 
Do you get "[Errno 13] Permission denied" message?
You have to resolve permission before baud rate.
Check the group of the dev file with the ls -l command above. Is it "dialout"?

steveb85, 2021-03-16:
I checked the group, it was root root, changed it to root dialout, but also put myself in every group possible (i know not good practice, but just trying to troubleshoot) and it didn’t see to help at all

volconst, 2021-03-17:
@steveb85 
- [ ] Do you get "[Errno 13] Permission denied" message?
- [ ] paste here the output of ls -l /dev/<your-device> here to see group permissions
- [ ] did you logout (or restart computer) after adding group?
- [ ] are you running security enhanced linux, what os are you running?

steveb85, 2021-04-20:
@volconst sorry I missed the notification for this. 

I did logout and restart, I am not running any enhanced security just 20.04 Ubuntu and I wasn’t getting permission denied message, i had similar problems with trying to run MatterControl on linux, regardless of baud rate etc, or permissions, I tried creating sym links to the port as well, but it just wouldn’t connect. I ended up installing Octopi, and that was able to automatically detect and connect instantly. It has to have something to do with the ports, as Octopi was able to see and connect via either the sym link or the direct port, but pronterface and MatterControl were both having trouble with that. Potentially a handshake issue as the fan started up for a bit before the connection failed? Well beyond my abilities at that point.

volconst, 2021-04-21:
Which version are you running?
Do you get any error messages? In console and log pane?
Enable Debug Communications item in the menu and paste the log here.
Why do you use a symlink?
Does the printer show some message on its display? It may halt due to thermistor problems, etc. Why would the fan stop? Is it stopped by a gcode command or marlin halt?

closed 2021-06-03 by DadOfThree