kliment/Printrun #1168: Can't connect to printer on Mac 10.5.6 Catalina

vintageb3, 2021-02-09:
Hi Folks. I've never used Printrun before and need to get it set up on my Mac to do esteps on a JGAURORA A1. The 2015 MacBook Pro can't see the com port...or at least I can't see how to point the app at the correct port. I've looked in System Report but can't get any clues. As above, I'm running 10.5.6 Catalina. On this Mac I also have El Capitan running on a different partition. Same problem there. On a MacBook Air, I have Seirra and Windows 10....same issue there....cant see the comports. Ive read on GitHub there were issues running Printrun on Catalina, but I downloaded someone's build that said he got it working. The app seems happy, but as I say, I can't find the port the printer is on. I read using /usbmodem was the correct port. Still can't see the printer. See screenshot. I hope someone can help me here. Thanks!

[image]

a2k-hanlon, 2021-02-14:
Hi @vintageb3, let me see if I can help you; I run Printrun on macOS 10.15.7.

First of all, please try the latest release from https://github.com/kliment/Printrun/releases/latest rather than a potentially out of date build from somewhere else. Version 2.0.0rc8 introduced several fixes relevant to the macOS build.

Next, `dev/usbmodem` is probably not quite what you need for the port. For example, my printer uses the port `/dev/tty.usbmodem123451` or `/dev/cu.usbmodem123451`. Also, I have an Arduino board with a CH340 that registers the ports `/dev/tty.usbserial-1420` and `/dev/cu.usbserial-1420`. Your printer will probably have something similar.

Make sure you power up your printer and connect it to your computer before you start Pronterface. Usually, Pronterface will present the serial ports available to it in the port dropdown. If it doesn't, you can determine what serial ports are available to your computer by opening a Terminal window and running the command `ls /dev/cu.* /dev/tty.*` and looking for something similar to the ports I mentioned above in the output.

If you don't see any ports like the above from the `ls` command when your printer is connected and powered on, then you may still be having an issue with drivers. In that case, https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all has very good info about CH340 drivers.

Please write back if you need any clarification, if you still have trouble after trying these suggestions, or if you successfully resolve the problem. Thanks!

vintageb3, 2021-02-14:
Hi There! Thanks so much for your reply and advice it’s greatly appreciated!  I will inwardly digest your advice and try out your suggestions in due course. I bought a Raspberry Pi4 and got Octoprint up and running to double check the USB port and chip. It seems the chip has issues. I’m currently working on fitting a BTT SKR-Mini E3 and display to this 3D printer, but I still want to get Printrun working. Please bear with me, I will get back to you when I get a chance to implement your advice. Thanks once more! Best! m

a2k-hanlon, 2022-04-19:
Hi @vintageb3, this conversation has become quite stale (no update in over a year), so I am going to close the issue now. If you need further help, please feel free to open a new issue. Cheers!

closed 2022-04-19 by a2k-hanlon