kliment/Printrun #1170: Locale problem stops pronterface from starting

Gearme, 2021-02-13:
I am getting the following stacktrace (and a `Failed to execute script pronterface` Popup) when trying to run pronterface-windows-x64-3.8.-2.0.0rc8:
```
$ ./pronterface.exe
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2455, in __init__
  File "printrun\pronterface.py", line 186, in __init__
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1699) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

```

DivingDuck, 2021-02-13:
Thanks for reporting. I had never seen this error on my Windows10 x64 system. I try to understand what is going on. Some questions:

What is your OS exactly and what is the language? 
Is there something special you did?
Did you modify a language file?
Where are your language files located

kliment, 2021-02-14:
I think this is related to https://discuss.wxpython.org/t/wxpython4-1-1-python3-8-locale-wxassertionerror/35168
We probably should drop back to wxPython 4.0.x for non-macos builds (including Windows) until this is fixed.

kliment, 2021-02-14:
@Gearme can you please check whether this build fixes the problem for you?: https://github.com/kliment/Printrun/suites/2036579493/artifacts/40949801

DivingDuck, 2021-02-14:
Interesting. Maybe worth a test with Py3.7 too?

DivingDuck, 2021-02-25:
@Gearme, did the provided fix from @kliment solve your problem?

Gearme, 2021-02-28:
Unfortunately not, it did not:
```$ ./pronterface.exe
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2455, in __init__
  File "printrun\pronterface.py", line 186, in __init__
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(LC_ALL, NULL), "C") == 0" failed at ..\..\src\common\intl.cpp(1579) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!```

DivingDuck, 2021-02-28:
Can you do me a favor? Can you download and run this version? In addition, what language version have your Windows 10 system?
https://github.com/DivingDuck/Printrun/suites/2142874027/artifacts/43836628

DivingDuck, 2023-02-12:
I know it was long time ago. We finally sort out all problems around the issue.

Here is a link to the latest Windows build.
https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark it as completed. Thanks for your report.

closed 2023-02-12 by DivingDuck