kliment/Printrun #1173: How are we in 2021 and we still need a serial connection?

serget2, 2021-02-20:
I only have Usb's, my connection to my Marlin card is with USB, Usb connects fine with USB, but now I need to install a serial port, so I can use a serial to usb converter to connect through USB? Isn't that kind of overkill? Am I doing this wrong? Is there a way to connect easily to USB like people in 2021 are used to? Or do I really need a timemachine and go back to the olden days where they never even heard of a 3D-printer to get a serial connection for my windows 10 PC?  I know you github guys like to brag about coding in commandlines and using old tech but it's more impressive if you get with the times and new tech guys

kliment, 2021-02-20:
I hope this is a joke, but in case it's not:

Most printer controller boards have a USB to serial converter on board. This can be embedded in the main microcontroller (so it just shows up as a virtual serial port), a separate microcontroller, or a dedicated chip like FT232 or CH340 or CP1201. To the operating system, all of those appear as a serial port (with the correct driver installed). You do not need a real serial port or a serial to usb converter, it's built into your printer electronics. Of course all of those connect via USB and we support that. If the port for your printer controller is not showing up, you are probably missing some drivers. What driver you need depends on your printer electronics and you should check if it has some instructions.

serget2, 2021-02-20:
It was not a joke, the only way I got it to work was by demolishing a raspberry device and turn it in to an octoprint and work like that

kliment, 2021-02-20:
You are almost certainly doing it wrong. That said, I don't think we can help you as your device configuration has nothing to do with this software.

closed 2021-02-20 by kliment

dlarue, 2021-02-20:
Serget2, USB is a serial interface, Ethernet is a serial interface, WIFI is a serial interface. SATA is a serial interface... Your problems have more to do with not understanding what this means and how your chosen operating system works, or doesn't work.