kliment/Printrun #1174: Packaging instruction issue on Windows

Henri-J-Norden, 2021-02-21:
On Windows 10 (insider preview build 21296) using PowerShell the packaging command listed in [buildinstructions.txt](https://github.com/kliment/Printrun/blob/master/buildinstructions.txt#L58) gives the following error: 
`pyi-makespec: error: argument --add-data: invalid add_data_or_binary value: 'images/*'`

The issue can be fixed by adding quotation marks around the add-data parameters, like so:
`pyi-makespec -F --add-data "images/*;images" --add-data "*.png;." --add-data "*.ico;." -w -i pronterface.ico pronterface.py`

However, I do not know if this change would cause issues in other environments (e.g. Linux).

DivingDuck, 2021-02-21:
@Henri-J-Norden , can you please try the batch file [release_windows.bat](https://github.com/kliment/Printrun/blob/master/release_windows.bat) instead? The batch include everything for building Pronterface on Windows 10 via command line from scratch. It will work best with latest Python 3.7 (which is actual 3.7.10).

Please let me know if it works for you too.

Henri-J-Norden, 2021-02-23:
@DivingDuck release_windows.bat does indeed succeed building with Python 3.8.5, even in PowerShell without the change mentioned in my original post, which is surprising.

_However, I still ended up using version 1.6, since unfortunately the ['Failed to execute script pronterface' issue](https://github.com/kliment/Printrun/issues/1006) was not resolved for me by building the code myself._

DivingDuck, 2021-02-23:
@Henri-J-Norden, thanks for your feedback.

Well, lets see what exactly the problem is. There should be more than only failed execution by something. I'm using Pronterface on regular base for my own printers with Windows10 w/o any script errors, so maybe there is some 'air' to improve your problem. 
As this is an different problem I would kindly ask you for answering the template below  (as RC5 is outdated and we are on RC8 or rather actual master branch) with suggested following information:

- An complete description what your problem is,
- Pronterface version, OS version and OS language, e.g.: Pronterface RC8, Windows 10 Pro x64 20H2, English
- What printer, gcode flavor, firmware version you are using etc (especially if it happen during printing,communication with the printer).
- When did it happen - e.g with starting Pronterface, running a print or what ever. 
- Add a screenshot and log file to it, printrunconf.ini is also helpful sometimes.
- ...

Simply everything what will/can help us understanding your problem better for helping you.

volconst, 2021-02-25:
> I should note, however, that I still ended up using version 1.6, since unfortunately the ['Failed to execute script pronterface' issue](https://github.com/kliment/Printrun/issues/1006) was not resolved for me by building the code myself.

@Henri-J-Norden , you can just start pronterface.py with the python interpreter (by double clicking?)

Henri-J-Norden, 2021-03-03:
> @Henri-J-Norden , you can just start pronterface.py with the python interpreter (by double clicking?)

When running pronterface.py in Python I only get the following error:
```WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (C:\Users\henri\Desktop\Progr\Printrun\printrun\__init__.py)
Traceback (most recent call last):
  File "C:\Users\henri\Desktop\Progr\Printrun\pronterface.py", line 62, in <module>
    app = PronterApp(False)
  File "C:\Users\henri\Desktop\Progr\Printrun\printrun\pronterface.py", line 2455, in __init__
    self.mainwindow = PronterWindow(self)
  File "C:\Users\henri\Desktop\Progr\Printrun\printrun\pronterface.py", line 186, in __init__
    self.SetIcon(wx.Icon(iconfile("pronterface.png"), wx.BITMAP_TYPE_PNG))
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1699) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!``` 

I tried:
1. Python 3.8.5 with both wxPython 4.1.0 and snapshot build 4.1.2a1.dev5103
2. Python 3.9.2 with wxPython 4.1.0 and snapshot builds 4.1.2a1.dev5103 and wxPython-pdb-4.1.2a1.dev5165
3. Python 3.7.9 with wxPython 4.1.0
but all of them result in the same error.

Henri-J-Norden, 2021-03-03:
@DivingDuck I pulled your master branch (cd8736f550eb55) and it runs fine in the interpreter without generating the error using Python 3.9.2 with wxPython 4.1.2a1.dev5165! Building with `release_windows.bat` worked with the following changes:
```diff
diff --git a/release_windows.bat b/release_windows.bat
index 41c366f..00f4db2 100644
--- a/release_windows.bat
+++ b/release_windows.bat
@@ -68,7 +68,7 @@ if exist v3 (
    echo ****** No virtual environment named v3 available                ******
    echo ****** Will create first a new virtual environment with name v3 ******
    echo **********************************************************************
-   py -3.7 -m venv v3
+   py -3.9 -m venv v3

    echo *********************************************
    echo ****** Activate virtual environment v3 ******
diff --git a/requirements.txt b/requirements.txt
index a3b2530..c793b35 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -1,5 +1,5 @@
 pyserial (>= 3.0)
-wxPython (== 4.1.0)
+wxPython (>= 4.1.0)
 numpy (>= 1.8.2)
 pyglet (>= 1.1)
 cffi
```
I also had to manually install wxPython-4.1.2a1.dev5165 into the v3 virtual env, otherwise I would still get the 'Failed to execute script pronterface' error when running the exe (with wxPython 4.1.1). I could not try 4.1.0, because it would fail installing with pip saying that I'm missing Visual C++ 14.2 build tools, which is not true.

DivingDuck, 2021-03-03:
This line tells me that something is wrong with your setup:
WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (C:\Users\henri\Desktop\Progr\Printrun\printrun\__init__.py)
Traceback (most recent call last):

I use VS2019 that provide also all compiler tools. Maybe worth to install the community version with complete C++ support

One remark when you deal with different python compilers on the same system. Each version need to have it own virtual environment or delete your old virtual enviroment build and also all  __pycache__ folders in all source code directories of Printrun. After this run release_windows.bat to build the hole environment new from scratch with py 3.7.9. In case you had install all python versions on your system, rename all other versions like 'Python38x' so that they don't impact the installation with wrong libraries (I mention this, because this happen to me last year and made me some headache before I understood what happened).

I still need to know what language your os is.

BTW, good catch for the development version of wxPython

Henri-J-Norden, 2021-03-03:
OS version is English and I do have Visual Studio 2019 with MSVC v142 x86 build tools and C++/CLI support for v142 installed, which is why it's also weird that wxPython 4.1.0 failed to install with the error message `distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.2 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/`. I suppose there might still be something wrong with it, such that Python is unable to figure out where my VS installation is.

I deleted the existing v3 environment I had (with 3.8.5) and created a new one using 3.9.2 for building. Before I tried building, I installed the requirements for each version with `%localappdata%\Programs\Python\Python<version>\python.exe -m pip install -r <Printrun_dir>\requirements.txt`, opened pronterface.py with the appropriate IDLE version and ran it from there.
I did not delete the __pycache__ dirs, but that doesn't seem to be necessary, as the cache files differentiate between version (e.g. `pronterface.cpython-38.pyc` vs `pronterface.cpython-39.pyc`). I only have one version of 3.7, 3.8 and 3.9 each, so they don't seem to conflict too much.

DivingDuck, 2021-03-03:
change this lines in my batch:
```
rem echo ***************************************************************
rem echo ****** Bug on wxPython 4.1.x workaround for Python 3.8.x ******
rem echo ***************************************************************
rem wxPython 4.1.1 cause a crash under Windows 10, Issue #1170
rem Relevant only in combination with Python 3.8.x.
rem Further information:
rem https://discuss.wxpython.org/t/wxpython4-1-1-python3-8-locale-wxassertionerror/35168
 pip uninstall wxPython
rem pip install wxPython>=4.0,<4.1
 pip install -U --pre -f https://wxpython.org/Phoenix/snapshot-builds/ wxPython
```
It will take care that you have the latest developer version what is actual wxPython  4.1.2a1.dev5165+64e5d863
For the moment I will not release this change as we usually don't want to deal with development versions in alpha state. Anyway I will adapt this change in my test branch and look for a VM with windows 10 in english.
BTW, is it 'en-us' ?

Henri-J-Norden, 2021-03-03:
> BTW, is it 'en-us' ?

Yes

DivingDuck, 2021-03-05:
Hi @Henri-J-Norden,
I now have a US version of Windows 10 and it seems I found an issue. Can you please make a test for me? Take one from the not working versions (RC8) and check if it is still not working. Then rename the config file `printrunconf.ini` in something different like `printrunconf.ini.sav`. This file can exist on two places:
```
c:\Users\YOUR_USER_NAME\AppData\Local\Printrun\Printrun\printrunconf.ini
c:\Users\YOUR_USER_NAME\printrunconf.ini
```

Please rename both and then try to start the non working pronterface.exe again. Please attach the old config file if Pronterface is starting now.

Thanks in advance, DD

Henri-J-Norden, 2021-03-16:
Hi, sorry for the delay.
Unfortunately, I renamed both the file under Appdata and in my user root dir and deleted c:\Users\YOUR_USER_NAME\printrunconf.ini~bak, but the RC8 executable still gives the 'Failed to execute script pronterface' error.

DivingDuck, 2021-03-17:
Thanks for your answer. Sorry, now I am out of ideas. All versions of wxpython 4.x are running on the newly created Windows10-x64 system with US language.
 
I saw this issue on the newly installed US version and I remember this behavior on my system in the past (just forgot it since then). The main driver was a incompatibility of config file between 1.9 and the 2.0 RC version when I often jump back and fore between the two versions. Some when the config file became unusable. I replicate this problem and the issue below both on the new created machine.
An other problem, that also force this error, was from a specified log file path when you enable logging where a path not exist, suffer of rights to write, is protected or include characters that are not allowed for windows path or file naming (e.g. when I copy a config file between computers or virtual machines). 
Anyway, this seems not to happen in your case and I can't replicate your problem. Happily it works for you with the developer version of wxPython 4.1.2a1.dev5165+64e5d863.

It is still a bit frustrating that it isn't possible to replicate the behavior, but so it is.

magnusjjj, 2021-08-25:
If you, in pronterface.py , do self.locale = wx.Locale(wx.LANGUAGE_ENGLISH)

instead of wx.Locale.GetSystemLanguage())

I debugged this for two hours yesterday. It seems to be an issue with libpng, where if the icons has a very very common issue, it tries to do something, print an error message or otherwise, and sets the locale. Apparantly, other people that have debugged this in other programs with wxpython has found that if you instead save it as a .bmp instead of .png it goes away. I didn't try it for printrun though.

Not.. the most pleasant, but at least it gets it working.

DivingDuck, 2021-08-25:
Well, this is a way, you can do it for a English version, but unfortunately there are also a lot users outside English spoken territories. 

In the end you don't need to modify any code, as it is a well known / reported bug in wx-python and the development team of wx python had solved it with a new development release in January 2021 (the one I had mentioned above). 
Using the latest development version of wxpython will fix the problem. Depending of the python version you are using you need also tho change the minimum wxpython in the file requrement.txt (we had change it back to an older version because of compatibility problems with other Linux distributions). 
For Python 3.9 you need to update by hand the entry in the file requrement.txt to minimum 4.1.1 as older versions will crash the compilation process with Python 3.9x for a Windows executable.

For ease of use you can copy the file release_windows.bat from my test branch  (https://github.com/DivingDuck/Printrun/tree/test) as my commitments are waiting for merging. Read the first page in the batch file to set up your python version for the script correctly (it is documented, just follow the instructions).

magnusjjj, 2021-08-25:
I mean, updating to the latest library was the *first* thing I tried. And other people have also been unsuccessful with it. I have the latest version right now, and changing back that line breaks everything.
I am in sweden, with a swedish locale. Like I said, it's not a pretty 'fix', in fact it's not a fix, more a tiny sliver of ducttape on a wound.

magnusjjj, 2021-08-25:
Nevermind, saw your pip upgrade. Going to test that.

magnusjjj, 2021-08-25:
Nope, that didn't work. Installed it with the same command as in your bat script, still broken. Did you test it recently?

DivingDuck, 2021-08-25:
Yes I did, I is running on my systems all the time and one or two weeks ago an other user here told me it is working for him too.
https://github.com/kliment/Printrun/issues/1208#issuecomment-897445598

magnusjjj, 2021-08-25:
*sighs* Again, I am an idiot. Upgraded to python 3.9, went through the release because for some reason it failed in actually creating the virtualenv, and the release script just assumed that it did and installed the packages globally :'). Very very strange. Can't actually reproduce.

But managed to build it, and it works, but only on 3.9. Blergh. Sorry for wasting your time by not reading properly :). Here's hoping your patches get merged.

magnusjjj, 2021-08-29:
It would, if it were reproducible. I redid my steps a couple of times and it never fails even with the automated script. I have no clue why it did it the first time. So, mark this as 'unable to reproduce' :')

DivingDuck, 2021-08-29:
Yes, we leave it as it is. Thanks for your feedback.

DivingDuck, 2023-02-12:
I know it was long time ago. We finally solved the issues around the windows installation here in the repository.

A short explanation for Windows packaging instructions can be found in section [Windows](https://github.com/kliment/Printrun#windows-1) at the main page of this repository.

Here is a link to the latest Windows build.
https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark it as completed

closed 2023-02-12 by DivingDuck