kliment/Printrun #1175: [SKR E3 Mini v2.0] Cant have the start commands

LordDarkneo, 2021-02-22:
Hello! 

I have been fighting with this for days now without finding a solution. I have downloaded the BTT Marlin frimware for the ENDER 3 and SKR E3 Mini v2.0, compiled it and copied it to the SD.
The printer seems to be all fine, but when I connect the printer through USB, I only have the message:
`Connecting...
Printer is now online.`
The commands are ok, but I cannot understand why I do not have the start command (rebooting the printer), as I normally have on my 8bits stock boards. Is this something to be fixed on Marlin side? on Pronterface? 
Thanks for your help!

a2k-hanlon, 2021-03-17:
Hi @LordDarkneo, I'm not sure exactly what you mean by start command. When Pronterface connects to a printer, it does not send any g-code command that would reset the printer. Do you need your printer to reset when Pronterface is connected to it? You said "commands are ok;" does that mean you are not having any problems operating your printer from Pronterface? Please provide some clarification.

LordDarkneo, 2021-03-17:
Hello,
Talking about stock boards, when I connect them through USB, the board "reboots" and I can see all the starting gcode (somehow an external reset?)
My main concern is to be able to debug the mmu connection when everything is powered up.
Regards

a2k-hanlon, 2021-03-17:
I did some more digging. Whether a board resets when a host (eg. pronterface) connects to it over serial depends on the hardware. On many Arduino boards, and other controllers using the same microcontrollers as Arduino boards, the hardware is designed with "reset on connect" functionality. However, the BTT SKR E3 Mini v2.0 is based on an STM32 microcontroller and evidently does not have this hardware. Therefore, it is normal for it not to reset when connecting to a host.

See https://reprap.org/forum/read.php?1,442013 and https://github.com/MarlinFirmware/Marlin/issues/4041.

So as far as I can tell, this is expected behaviour and it is not related to Pronterface. I don't think this will cause you any problems.

a2k-hanlon, 2021-03-17:
Does that resolve your question?

LordDarkneo, 2021-03-17:
Thanks a lot for your answers on this! That's pretty clear to me, and yes it fully answers the question! This behavior is hardware related then, I'll try to find a way to access what I am looking for another way.

Thanks for your time and answer once again, have a great day

closed 2021-03-17 by LordDarkneo

volconst, 2021-03-17:
@LordDarkneo see https://github.com/kliment/Printrun/issues/1102#issuecomment-688481295

LordDarkneo, 2021-03-18:
@volconst DTR is checked. My issue is not that I cannot to the board, more that there is no "reset on connection" function, that does not allow to track the initialization of the board.
On my Créality stock board (ATMega):
`External Reset
Marlin 2.0.7.2

echo: Last Updated: 2020-10-15 | Author: darkneo, CR10-V2)
echo:Compiled: Mar  5 2021
echo: Free Memory: 1867  PlannerBufferBytes: 1472
DIGIPOTS Loading
DIGIPOTS Loaded
DIGIPOTS Loading
DIGIPOTS Loaded
echo:V82 stored settings retrieved (721 bytes; crc 45666)
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.19 Z401.00 E417.95
echo:; Maximum feedrates (units/s):
echo:  M203 X750.00 Y750.00 Z15.00 E75.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X2000.00 Y2000.00 Z100.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P800.00 R1000.00 T800.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 W I0 J0 Z-0.38486
echo:  G29 W I1 J0 Z-0.17289
echo:  G29 W I2 J0 Z0.10890
echo:  G29 W I3 J0 Z0.45304
echo:  G29 W I4 J0 Z0.78870
echo:  G29 W I0 J1 Z-0.38885
echo:  G29 W I1 J1 Z-0.20082
echo:  G29 W I2 J1 Z0.09444
echo:  G29 W I3 J1 Z0.40965
echo:  G29 W I4 J1 Z0.70142
echo:  G29 W I0 J2 Z-0.30107
echo:  G29 W I1 J2 Z-0.15693
echo:  G29 W I2 J2 Z0.07200
echo:  G29 W I3 J2 Z0.34132
echo:  G29 W I4 J2 Z0.59219
echo:  G29 W I0 J3 Z-0.15045
echo:  G29 W I1 J3 Z-0.08711
echo:  G29 W I2 J3 Z0.09893
echo:  G29 W I3 J3 Z0.34431
echo:  G29 W I4 J3 Z0.56676
echo:  G29 W I0 J4 Z-0.03574
echo:  G29 W I1 J4 Z-0.01080
echo:  G29 W I2 J4 Z0.18471
echo:  G29 W I3 J4 Z0.42262
echo:  G29 W I4 J4 Z0.69344
echo:; Material heatup parameters:
echo:  M145 S0 H205 B60 F0
echo:  M145 S1 H230 B80 F0
echo:; PID settings:
echo:  M301 P18.45 I1.39 D61.23
echo:  M304 P690.34 I111.47 D1068.83
echo:; Z-Probe Offset (mm):
echo:  M851 X47.00 Y0.00 Z-0.67
echo:; Filament load/unload lengths:
echo:  M603 L430.00 U435.00
echo:; Filament runout sensor:
echo:  M412 S1 D5.00
echo:SD card ok`

On my SKR (STM32):
`Connecting... 
Printer is now online.`

volconst, 2021-03-18:
What if you reset the printer pressing its reset switch if available?
Or power off printer, click connect in Pronterface and power on printer.