kliment/Printrun #1176: Pronterface (current release) is infected with maleware.

JerryDennisEasley, 2021-03-01:


kliment, 2021-03-01:
It is not. See #1159 - it's a false positive

JerryDennisEasley, 2021-03-01:
[image]

kliment, 2021-03-01:
Please use the "report as false positive" option if you can. It's being detected because it has a python interpreter in it, which several forms of malware also do.

kliment, 2021-03-01:
Apparently the same thing happens quite regularly - [here](https://steamcommunity.com/app/379210/discussions/0/353915953245690255/)'s an example of one, [here](https://community.adobe.com/t5/download-install/virus-in-installation-file/td-p/9369736) is another

BurtHarris, 2021-03-03:
I also encountered this.   Note: Windows Defender doesn't seem to have a "report as false positive" option.   I'll try to dig a little deeper on that.

BurtHarris, 2021-03-03:
Here's a link [to submit an antivirus false positive report to Microsoft](https://www.microsoft.com/en-us/wdsi/filesubmission).    It would probably be more effective if one of the developers submitted it.

DivingDuck, 2021-03-03:
No need for doing this. Windows will send an false positive if you allow to send a notice. In general nearly each new program have this problem if it is not known as new version - and even if a company have send out a report in advanced.
BTW:
I never got a warning from AV for Pronterface, and this is why:
https://www.virustotal.com/gui/file/3d6a2ed084e6d222c9e7e7613bfd19b3930962634de36bf87b8f78cee91e7f7a/detection

There are several free services on internet available to check for false positive behaviors. Just use them. This will help most to get good maintained and up to date databases for AV programs.

jorgegrillo68, 2025-03-30:
Hola estamos en el 2025 y sigue apareciendo la misma advertencia.

rockstorm101, 2025-04-05:
Hi @jorgegrillo68, as far as I know, there's nothing we can do. This is a false positive for which the antivirus software is responsible. If you have any ideas regarding how to prevent this from happening please let us know.